microsoft/AirSim #3098: how to record image at flying a multirotor simultaneously

lucafei, 2020-10-24:
hello, everyone. i want to recording the scene image and segmentation image at flying simultaneously. But when i executed the following codes, it always stay at recording image and can not move to desired position.
def record_image(multirotorclient):
    image_id = 0
    while True:
        responses = multirotorclient.simGetImages([
            airsim.ImageRequest("0", airsim.ImageType.Segmentation),  # depth visualization image
            airsim.ImageRequest("1", airsim.ImageType.Scene)])  # scene vision image in png format
        print('Retrieved images: %d' % len(responses))

        img_tmp_dir = os.path.join(os.getcwd(), "airsim_drone/image")
        seg_tmp_dir = os.path.join(os.getcwd(), "airsim_drone/segmentation")

        print("Saving images to %s" % img_tmp_dir)
        print("saving segmentation to %s" % seg_tmp_dir)
        try:
            os.makedirs(img_tmp_dir)
            os.makedirs(seg_tmp_dir)
        except OSError:
            if not os.path.isdir(img_tmp_dir) or not os.path.isdir(seg_tmp_dir):
                raise
        img_name = os.path.join(img_tmp_dir, str(image_id).zfill(4))
        seg_name = os.path.join(seg_tmp_dir, str(image_id).zfill(4))
        for idx, response in enumerate(responses):
            print('image type %s' % response.image_type)
            if response.image_type == airsim.ImageType.Segmentation:
                    print("Type %d, size %d" % (response.image_type, len(response.image_data_uint8)))
                    airsim.write_file(os.path.normpath(seg_name + '.png'), response.image_data_uint8)
            elif response.image_type == airsim.ImageType.Scene:
              response.compress:  # png format
                   print("Type %d, size %d" % (response.image_type, len(response.image_data_uint8)))
                   airsim.write_file(os.path.normpath(img_name + '.png'), response.image_data_uint8)
        image_id += 1
        time.sleep(0.2) #fps=5hz 

def main():
    client = airsim.MultirotorClient(ip)
    client.confirmConnection()
    client.enableApiControl(True)
    client.armDisarm(True)
    mythread = threading.Thread(target=record_image, args=(client,))
    mythread.start()
    client.moveToPositionAsync(-150, 0, -0.5, 3).join()
    mythread.join()

do you have any idea for that? how to modify this programm?

rajat2004, 2020-10-28:
Is it not moving at all, as in maybe not receving the command? Maybe try with different clients for different threads rather than a single client

LeaRain, 2021-04-07:
Hi,

I don't know, if this is still relevant for you, but maybe someone else finds this issue and needs help. So, I removed your separate threading and the join in moveToPositionAsync(), because join() is waiting for the task to complete. I also added a takeoffAsync().join(), so the drone is in position for further moving to another position:

    def main():        
        client = airsim.MultirotorClient()   
        client.confirmConnection()
        client.enableApiControl(True)
        client.armDisarm(True)
        client.takeoffAsync().join()
        client.moveToPositionAsync(-150, 0, -0.5, 3)
        record_image(client)

Removing the extra threading may not be the best or cleanest solution, but it works (on my machine).