microsoft/AirSim #3099: how to set the color of segmentation image

lucafei, 2020-10-25:
i want to get and set the color of segmentation image. Do you know how to set the color of segmentation parts and get the color of  segmentation, because now the image api don't provide function to deal with this. It seems like the color to segmenation is random selected.

GimpelZhang, 2020-10-26:
Haven't tried, but it seems that there is a part in the introduction document about this:

[Segmentation](https://microsoft.github.io/AirSim/image_apis/#segmentation)

jonyMarino, 2020-11-04:
Hi @lucafei! do you think that the color assigned by seg_rgbs.txt is not enough? (I know you've posted an issue about that)

lucafei, 2020-11-05:
> Hi @lucafei! do you think that the color assigned by seg_rgbs.txt is not enough? (I know you've posted an issue about that)

you can get and set segmentation color using https://github.com/microsoft/AirSim/blob/d59ceb7f63878f5e087ea802d603ba0fd282ff56/PythonClient/airsim/client.py#L425
and https://github.com/microsoft/AirSim/blob/d59ceb7f63878f5e087ea802d603ba0fd282ff56/PythonClient/airsim/client.py#L443. 
but you have to download the lastest version of airsim. Although it works to set and get the segmenation color, it has problem at  outputing color. I find when i get the output segementation image, this image has different color from what you get or set using API, which even doesn't exit in seg_rgbs.txt.

jonyMarino, 2020-11-05:
Thanks for your explanation.
So let's continue with #3103 alone.

closed 2020-11-12 by jonyMarino