microsoft/AirSim #3101: Setting up Unreal Environment for Linux?

yx0123, 2020-10-27:
I want to run my project on Linux and I am trying to add the AirSim plugin to the Landscape Mountains by following the steps [here](https://github.com/microsoft/AirSim/blob/master/docs/unreal_custenv.md). Since there is no Epic Games Launcher for Linux, I downloaded the environment using Windows. I copied the environment folder to my Linux Machine but Step 4 fails because there is no Visual Studio available for Linux. Do I have to build UE and AirSim on my Windows  to set up the Environment for running in Linux? Or is there any way to add AirSim plugin without using Visual Studios?

GimpelZhang, 2020-10-27:
From my experience, first you need a linux version UE4 and a linux version AirSim, which both work well in your linux system. 

Then in windows, follow the steps in the link you provided to make the custom environment ready. 

Copy your custom environment folder to the linux platform, and copy the AirSim plugin folder (linux version) to your project, just like in Step 5 in your link, to replace the windows one. 

Use the UE editor on linux to open the environment, hopefully things will work.

yx0123, 2020-10-27:
> From my experience, first you need a linux version UE4 and a linux version AirSim, which both work well in your linux system.
>
> Then in windows, follow the steps in the link you provided to make the [...]

Thanks for your reply! Does this mean that I skip Step 5 in Windows?

yx0123, 2020-10-27:
> From my experience, first you need a linux version UE4 and a linux version AirSim, which both work well in your linux system.
>
> Then in windows, follow the steps in the link you provided to make the [...]

Does this mean I also need to install AirSim and UE4 on Windows for the steps to work?

GimpelZhang, 2020-10-27:
> Does this mean I also need to install AirSim and UE4 on Windows for the steps to work?

I think so. Or at least in my situation I have them both on my Windows.

closed 2020-10-28 by yx0123