microsoft/AirSim #3103: problem at function simSetSegmentationObjectID

lucafei, 2020-10-27:
hello, guys, i'm using API simSetSegmentationObjectID to change the color of segmentation. But i find when i used simSetSegmentationObjectID to assign a specific value to a specific mesh, it actually changed the value of mesh. like following:
success = client.simSetSegmentationObjectID('runway[\w]*',19,True)
i can clearly see the color is already changed.
after i save the segmentation image. i open this image and read the pixel value. But i find it is not same with the rgb, which maps to value 19: you can find the RGB values for each object ID in this file https://microsoft.github.io/AirSim/seg_rgbs.txt. In my case i read the RGB value of saved segmentation image as [0,126,138] , but i set the value as 19 which maps to RGB value as [0, 53, 65]. It makes me very confused. Do you guys help me and give me some tipps? Thank you in advance.

rajat2004, 2020-10-28:
Are you using a release binary or the latest master? There's a PR https://github.com/microsoft/AirSim/pull/2845  in master currently which fixes color corrections and makes the ids match with the values

lucafei, 2020-10-31:
> Are you using a release binary or the latest master? There's a PR #2845 in master currently which fixes color corrections and makes the ids match with the values

yes, i have built the latest version of airsim from master. And the simSetSegmentationObjectID and simGetSegmentationObjectID works. But when i read the segmentation image using PIL and get the RGB value. I found it is different from what i used SetSegmentationObjectID to set. So i don't know where is the problem? maybe the file https://microsoft.github.io/AirSim/seg_rgbs.txt is updated? because the RGB value derived from this file is always different from the final saved RGB value.

chickenbestlover, 2021-01-05:
Hello, the problem happened to me as well. Did you find any solutions to this?