microsoft/AirSim #3104: PX4 SITL with MAVROS

lberridge1992, 2020-10-27:
Hello,

I'm trying to setup PX4 SITL using MAVROS as the main mechanism of sending offboard API commands. I've got 2 VMs as follows (apologies for concealing the IP addresses as I'm on a corporate network):
- One Windows VM running Airsim and PX4 SITL (IP address ending 86)
- One Linux VM running ROS which I'm using to represent my offboard controller and use MAVROS to communicate with PX4 (IP address ending 118)

Launching PX4 SITL with airsim yields the following:
[image]

Then on my ROS VM I'm running roslaunch mavros px4.launch fcu_url:=udp://XXX.XX.XX.127:14540@XXX.XX.XX.86:14580 but the mavros node doesn't seem to connect and I can't send the arm/takeoff command by triggering the appropriate ROS service:

[image]

I've used the IP address ending in 127 in fcu_url as it is listed as the broadcast IP in the log from PX4 but I've also tried ending 86 too along with about a million combinations of IP addresses and ports. I wasn't sure if it wasn't working because Airsim is already using some of the ports required for MAVROS.

Is it possible to configure Airsim, PX4 SITL and MAVROS to work as described above and would anyone be able to advise on how I could get this working?

rajat2004, 2020-10-28:
Issue https://github.com/microsoft/AirSim/issues/2954 might be useful, there's something about MAV_BROADCAST mentioned there which might be required. I would also suggest to first make sure that individual components are working correctly, such as AirSim+PX4 SITL in VM, then PX4 in VM + Mavros, and then add AirSim into the equation

lberridge1992, 2020-10-29:
Thanks for the response. I've moved across to a local machine to double check that all the individual elements are working. The connection between AirSim and PX4 seems quite intermittent. Sometimes I get a successful connection at the start and other times my drone starts underneath the ground plane and just falls. Most of the time, if I don't get this I can use the Python API to fly the drone in an orbit but it mostly orbits for a little bit then falls out of the sky. The message "timeout on HilActuatorControlsMessage" is printed in the AirSim log. I had a look into this and it looks like there was a fix to a similar issue added in #2556 but then later removed in commit 3fdc5a9b0cb85dc79953aed00325aab90af90e01. Is anyone else seeing similar issues with PX4 v1.10.1 and the latest master branch of AirSim? The only thing I could think of that I'm doing that might be different to most is using Unity as the sim rather than Unreal (I can't use Unreal on my system even just to test that it works)

lberridge1992, 2020-11-16:
Wanted to post an update on this so far as I've done a bit more investigation:
- I've tested the config of PX4 with the Unreal sim and I do not see the same behaviour with the drone falling out of the sky as I do with the Unity sim
- I've tried disabling lockstep mode which works better and removes the "timeout on HilActuatorControlsMessage" messages for obvious reasons. However, I still see the drone dropping out of the sky when it enters a failsafe mode (see image)
[image]
I've added some additional log messages to report when the data is stale and it appears to be that the global position, local position and local velocity are all being classified as stale by PX4. Again this only occurs when using Unity.
- I implemented some of the changes described in the PR associated with #3126 as both of those queries could have been causing my problem and I believe the fixes are sensible. This hasn't resolved my failsafe issue but has fixed something else I was seeing a lot where the sim would initialise randomly falling through the air rather than in it's start location.

Does anyone have any thoughts on why the data observed by PX4 would be stale?

lberridge1992, 2020-11-19:
Bit more information on this that I've posted on the PX4 github page (https://github.com/PX4/PX4-Autopilot/issues/16234) as I believe it is caused by an issue in the firmware around polling (the destruction of a semaphore seems to stall the sensor module so it doesn't ever publish the topic within PX4). Whether that is triggered by Airsim I'm not sure but the fact that I only see it with the Unity binary and not Unreal is confusing. HITL works completely fine with Unity and I still do get "timeout on HilActuatorControlsMessages" in Unreal running SITL so have put it down to better error handling in the Unreal at the moment that just means it doesn't drop out of the sky when this occurs.

jonyMarino, 2021-06-03:
Drone falling from the sky was fixed in [the latest version (1.5.0)](https://github.com/microsoft/AirSim/releases)
Closed due to inactivity. Please, feel free to ask for it to be reopened if you wish to continue posting.

closed 2021-06-03 by jonyMarino