microsoft/AirSim #3105: How to get the source code of unreal engine?

DongXingshuai, 2020-10-28:
Hi, I want to build AirSim on my computer following:

# go to the folder where you clone GitHub projects
git clone -b 4.24 https://github.com/EpicGames/UnrealEngine.git
cd UnrealEngine
./Setup.sh
./GenerateProjectFiles.sh
make

Unfortunately, it reminded me that "fatal: repository 'https://github.com/EpicGames/UnrealEngine.git/' not found
". 

I have registered with Epic Games and linked my with Epic Games account to my Github account.

So, how to get the source code of unreal engine? Thank you very much.

xwnb, 2020-10-28:
You should join the UE project, and then you can fork repo to get the source code. For more, see [Downloading Unreal Engine Source Code](https://docs.unrealengine.com/en-US/GettingStarted/DownloadingUnrealEngine/index.html)

jonyMarino, 2020-11-04:
Hi @DongXingshuai, and welcome! This is not an airsim issue. Please ask in UE forums. Regards

closed 2020-11-04 by jonyMarino