microsoft/AirSim #3106: unexpected behaviour by moveToPositionAsync API

cdbharath, 2020-10-28:
I am trying a path planner algorithm for drones in Airsim. When I tried moveToPositionAsync API. I found that the drone doesn't move when the set point relative to the current position is less than 0.8. Is this the actual behaviour of the API?

Simple Test Case:

  # current x = 1
  # current y = 1 
  # current z = -1
  
  x = 1.2
  y = 1
  z = -1
  self.xy_velocity = 0.3
  moveToPositionAsync(x, y, z, self.xy_velocity, vehicle_name=self.vehicle_name)

Simulation Editor: Unity
OS: Linux

zimmy87, 2020-11-19:
Yes this is the intended behavior of the API. See https://microsoft.github.io/AirSim/apis/#lookahead-and-adaptive_lookahead for the lookahead and adaptive_lookahead parameters. These will need to be adjusted to see the desired behavior, but there is a minimum value for lookahead of 0.5 meters. See https://github.com/microsoft/AirSim/blob/e1de671eb12ba6d0f7483175fcf70ba8b176ef40/AirLib/src/vehicles/multirotor/api/MultirotorApiBase.cpp#L213-L227

closed 2021-01-04 by jonyMarino

Cming7, 2022-01-15:
> Yes this is the intended behavior of the API. See https://microsoft.github.io/AirSim/apis/#lookahead-and-adaptive_lookahead for the lookahead and adaptive_lookahead parameters. These will need to be [...]

Hello, can I change the minimum value?

Cming7, 2022-01-15:
> I am trying a path planner algorithm for drones in Airsim. When I tried moveToPositionAsync API. I found that the drone doesn't move when the set point relative to the current position is less than [...]

Excuse me, have you solved the problem yet? Can I change the minimum value?