microsoft/AirSim #3107: Question : I wanna get energy consumption

4iue0, 2020-10-28:
https://github.com/srivatsankrishnan/AirSim-1

I found it to get energy consumption in 'getTripStats()'.
But I cannot execute hello_drone.py with this error code.

WARNING:tornado.general:Connect error on fd 320: WSAETIMEDOUT

Plz, tell me how to use getTripStats().

rajat2004, 2020-10-28:
Maybe AirSim hasn't been started before running the script, or might be firewall or IP issues if running on different systems, etc.
Also, the issue should be opened against that repository

4iue0, 2020-10-29:
Thank you for answering.
I will try it.
Sorry, I wanna open issue in that repository, but its item of issue was close by owner.
I can only open issue here.

jonyMarino, 2020-11-04:
Hi @4iue0! How did it go?

closed 2020-12-11 by jonyMarino