microsoft/AirSim #3108: Blocks Coordination in Blocks Environment

Moj-Dev, 2020-10-29:
I want to define some random position points in the Blocks environment and train a drone as an RL agent to reach to the goals. I should reject the goals which are in the object's mesh area. How can I check it and choose just the points that are not in the blocks?

GimpelZhang, 2020-10-29:
Sizes and positions of all obstacles in the environment can be read through UE editor. So I think obtaining these information by hand can definitely solve this problem.

Moj-Dev, 2020-10-29:
@GimpelZhang Thanks for your comment. 
For each episode (Run), I want to randomize the position of objects. So I need a command to get the position of object mesh in the environment.

jonyMarino, 2020-11-13:
Hi! did you try with simSetObjectPose?

closed 2021-08-19 by jonyMarino