microsoft/AirSim #3109: I want to get high resolution images.

4iue0, 2020-10-29:
I changed settings.json file to get 720P images.
but I cannot get  it.
please tell me how to change settings to get 720P images.
`{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings_json.md",
  "SettingsVersion": 1.2,

  "SimMode": "Multirotor",
  "CameraDefaults": {
    "CaptureSetting": [
      {
        "Width": 1280,
        "Height": 720,
        "FOV_Degrees": 90
      }
    ]
  }
}`

OS:windows10
AirSim ver : 1.3.1

AIKUUM, 2020-10-29:
> I changed settings.json file to get 720P images.
> but I cannot get it.
> please tell me how to change settings to get 720P images.
>
> ```
> {
>   "SeeDocsAt": [...]

@4iue0, I will let you know if I set image resolution in "settings.json" file in future.
If possible can you give me a hint regarding how to proceed with the issue as in the link below:-https://github.com/microsoft/AirSim/issues/3078#issue-720046903

AIKUUM, 2020-10-29:
> {
>   "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings_json.md",
>   "SettingsVersion": 1.2,
> 
>   "SimMode": "Multirotor",
>   "CameraDefaults": {
>     "CaptureSetting": [
>       {
>         "Width": 1280,
>         "Height": 720,
>         "FOV_Degrees": 90
>       }
>     ]
>   }
> }

@4iue0, I used the same configuration as that of yours, but instead for a car. But still lower resolution (256*144) images are getting captured.

4iue0, 2020-10-29:
@AIKUUM 
U are right. Firstly I said that I cannot get 720P images. So, I'm trouble.

AIKUUM, 2020-10-29:
@4iue0, Was your drone simulation manual or automated??

4iue0, 2020-10-29:
@AIKUUM 
maybe, manual

AIKUUM, 2020-10-29:
> @AIKUUM
> maybe, manual

Did you try automation for drone simulation. For example did you try executing "hello_drone.py" file present in AirSim package??
Link for "hello_drone.py": https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/hello_drone.py

4iue0, 2020-10-30:
@AIKUUM 
do u understand my trouble? 
I already have known automated operation and have already used hello_drone.py.
 I don't have any problem in first step like hello_drone.py.
I wanna only get 720P images.

AIKUUM, 2020-10-30:
> @AIKUUM
> do u understand my trouble?
> I already have known automated operation and have already used hello_drone.py.
> I don't have any problem in first step like hello_drone.py.
> I wanna only get 720P images.

@4iue0, I understood your issue when had a look at just your title. I already told you that even I'm getting lower resolution (256*144) images for the same configuration in settings.json as that of yours. I will let you know If I can get higher resolution images somehow when I execute scripts and will let you know what settings/changes in the scripts would fetch higher resolution images. 

At the same time I also wanted you to help me with the issue if possible as in the link:
https://github.com/microsoft/AirSim/issues/3078#issue-720046903
Anyways I have resolved the issue I had.
Thank you

AIKUUM, 2020-11-03:
@rajat2004, can you please help us on how to capture higher resolution images? Even with the configuration as below in settings.json, we cant capture images based on the desired resolution:
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings_json.md",
  "SettingsVersion": 1.2,

  "SimMode": "Multirotor",
  "CameraDefaults": {
    "CaptureSetting": [
      {
        "Width": 1280,
        "Height": 720,
        "FOV_Degrees": 90
      }
    ]
  }
}
I'm able to get the default resolution (256*144). Can you suggest us how to change the resolution?

4iue0, 2020-11-04:
> > @AIKUUM
> > do u understand my trouble?
> > I already have known automated operation and have already used hello_drone.py.
> > I don't have any problem in first step like hello_drone.py.
> > I wanna only get 720P images.
> 
> @4iue0, I understood your issue when I had a look at just your title. I already told you that even I'm getting lower resolution (256*144) images for the same configuration in settings.json as that of yours. I will let you know If I can get higher resolution images somehow when I execute scripts and will let you know what settings/changes in the scripts would fetch higher resolution images.
> 
> At the same time I also wanted you to help me with the issue if possible as in the link:
> [#3078 (comment)](https://github.com/microsoft/AirSim/issues/3078#issue-720046903)
> Anyways I have resolved the issue I had.
> Thank you
@AIKUUM 
Sorry, I just misunderstood. I'm glad you to solve your problem.

AIKUUM, 2020-11-04:
@4iue0, Are you getting higher resolution images now?

rajat2004, 2020-11-04:
I think you need to add `ImageType` element in the `CaptureSettings`
You can also see [high_res_camera.py](https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/high_res_camera.py) for another example

AIKUUM, 2020-11-05:
> I think you need to add `ImageType` element in the `CaptureSettings`
> You can also see [high_res_camera.py](https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/high_res_camera.py) for another example

@rajat2004, For the single drone configuration in "settings.json", the simulation is successful for me. Can you tell me what the configuration should be for 2 drones in "settings.json"? Below is the configuration that I have used for 2 drones:
`
{
    "SettingsVersion": 1.2,
    "SimMode": "Multirotor",
    "Vehicles" : {
        "Drone1" : {
            "VehicleType" : "SimpleFlight",
            "AutoCreate" : true,
            "Cameras" : {
                "high_res": {
                    "CaptureSettings" : [
                        {
                            "ImageType" : 0,
                            "Width" : 4320,
                            "Height" : 2160
                        }
                    ],
                    "X": 0, "Y": 0, "Z": 0,
                    "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
                },
                "low_res": {
                    "CaptureSettings" : [
                        {
                            "ImageType" : 0,
                            "Width" : 256,
                            "Height" : 144
                        }
                    ],
                    "X": 0, "Y": 0, "Z": 0,
                    "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
                }
            }
        },
		"Drone2" : {
            "VehicleType" : "SimpleFlight",
            "AutoCreate" : true,
            "Cameras" : {
                "high_res": {
                    "CaptureSettings" : [
                        {
                            "ImageType" : 0,
                            "Width" : 4320,
                            "Height" : 2160
                        }
                    ],
                    "X": 1, "Y": 0, "Z": 0,
                    "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
                },
                "low_res": {
                    "CaptureSettings" : [
                        {
                            "ImageType" : 0,
                            "Width" : 256,
                            "Height" : 144
                        }
                    ],
                    "X": 1, "Y": 0, "Z": 0,
                    "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
                }
            }
        }
    }
}
`

Even with this cofiguration, 2 drones coincide. Morover when i execute a code for 2 drones, the execution is not successful. Can you please help me to proceed?

Ryton, 2021-04-13:
Actually, the json is very sensitive to correct naming ...
a pitty he just ignores it, instead of spitting an error, while for every comma or bracket that is wrong, the program will not run.

Can you try
			"CaptureSetting**s**": [ instead of 			"CaptureSetting":

That should work (did for me at least)!

closed 2022-01-12 by jonyMarino