microsoft/AirSim #3113: Initial Takeoff by drone(s)

AIKUUM, 2020-10-31:
Automated Drone simulation successfully completes when one drone is used (i.e execution of "hello_drone.py") even when initial takeoff by the drone is skipped.

But automated drone simulation of multi drones doesn't get completed when multiple drones are used (i.e execution of "multi_agent_drone.py") when initial takeoffs by the drones are skipped.

Can anyone tell the reason behind this difference?

jonyMarino, 2020-11-13:
Hi @AIKUUM! I don't understand your question. What do you refer to when saying, "when initial takeoffs by the drones are skipped."? do you comment that line in the script?

AIKUUM, 2020-11-16:
> Hi @AIKUUM! I don't understand your question. What do you refer to when saying, "when initial takeoffs by the drones are skipped."? do you comment that line in the script?

Yes, I tried commenting that line in the script.