microsoft/AirSim #3123: I want to know how to record without imageFile.

kensuke1921, 2020-11-03:
I use record button in AirSim because I want to record speed of the car and car position. 
However, record button ain AirSim also record the ImegeFile. So processing while driving is heavy.  
Please teach me how to record without imageFile.

tharindurmt, 2020-11-03:
You have to remove any elements under tag "recording" in the settings.json file.
Ex: I have the following section in my settings file. If I remove the it, AirSim will no longer record the images. 

```"Recording": {
    "RecordOnMove": true,
    "RecordInterval": 1,
    "Cameras": [
      {
        "CameraName": "FrontLeftCam",
        "ImageType": 0,
        "PixelsAsFloat": false,
        "Compress": true
      },
      {
        "CameraName": "FrontRightCam",
        "ImageType": 0,
        "PixelsAsFloat": false,
        "Compress": true
      },
      {
        "CameraName": "FrontCentreCam",
        "ImageType": 0,
        "PixelsAsFloat": false,
        "Compress": true
      },
      {
        "CameraName": "RearCentreCam",
        "ImageType": 0,
        "PixelsAsFloat": false,
        "Compress": true
      }
    ]
  }```

kensuke1921, 2020-11-13:
Thank you for advice.
It means like this picture？
Originally no code was written in settings.json.
So I wrote "recording".
But recording with imageFile is continuing...
Please teach me...
[image]

tharindurmt, 2020-11-16:
Try removing the  `Recording` tag altogether.
Do you have any content in your AirSim plugin? (I'm referring to any image recording code)

kensuke1921, 2020-11-17:
I removed the Recording tag altogether like this.
But recording with imageFile is continuing....
I think RecordingFile.cpp is seems to be related to this problem.

[image]

tharindurmt, 2020-11-18:
Yeah, this seems to be the default behaviour. Sorry for not catching it before.
What you can do is to comment out [these lines](https://github.com/microsoft/AirSim/blob/37a0aaebc171c5c7fa3be739edb74c1ae8e06807/Unreal/Plugins/AirSim/Source/Recording/RecordingThread.cpp#L121-#L124), so that the images are not captured. I haven't tried this myself. But commenting out these line should stop recording the images, if that's what you want. However, you'll most probably have to do some extra modifications to the /Unreal/Plugins/AirSim/Source/Recording/RecordingFile.cpp file by removing statements related to image capture, such as below.
```        std::stringstream image_file_name;
        image_file_name << "img_" << response.camera_name << "_" <<
            common_utils::Utils::toNumeric(response.image_type) << "_" <<
            common_utils::Utils::getTimeSinceEpochNanos() << 
            (response.pixels_as_float ? ".pfm" : ".png");```

kensuke1921, 2020-11-18:
Thank you for advice.
I comment out the line you specified.
But, all items are no longer recorded. For example, car position, car speed.
I want to know how to record except for only imageFile. 
Sorry.......

tharindurmt, 2020-11-19:
Can you paste the code you changed here?
Unfortunately, I won't have enough time to test this myself.

kensuke1921, 2020-11-20:
Thank you for reply.
I comment out the line you specified.
I only did it.
I can't understand how to change the other code.
Sorry..
[image]
[image]

tharindurmt, 2020-11-23:
Sorry, missed the email notification.
Did it throw any errors? Any log messages?

kensuke1921, 2020-11-23:
It didn't throw any errors and log message is nothing.

tharindurmt, 2020-11-24:
I'll try this myself. But will need some time as I'm in the middle of developing another AirSim application.

kensuke1921, 2020-11-25:
Thank you so much!

PR #2861 (2021-03-01): Extend Recording to multiple vehicles
Replaces #2847 (branch got messed up)

Closes #3029
Closes #3123
Closes #3214

Other related issues - #2307

New Additions -

1. Info of all vehicles is recorded (with additional column `VehicleName`), `VehicleName` parameter in `Cameras` field to specify specific vehicle to use the camera from
2. `Folder` field in settings to change the directory to store data in
3. Allow Recording with no images (just pose, etc data) (#3123)
4. `Enabled` parameter for starting recording when AirSim starts without needing to press R or call API (#3029) (Any naming suggest [... cut]
closed 2021-03-01 by zimmy87