microsoft/AirSim #3127: Update render functions called twice per tick

ykeuter, 2020-11-04:
`updateRenderedState `and `updateRendering `are both called twice per tick, in:
https://github.com/microsoft/AirSim/blob/040a08ae6e789a6ad952837acf8f8323a1649819/Unity/AirLibWrapper/AirsimWrapper/Source/PawnSimApi.cpp#L57
and
https://github.com/microsoft/AirSim/blob/67d5b6af4ae317d796b17abc95e3f478370aefa7/Unity/AirLibWrapper/AirsimWrapper/Source/SimMode/SimModeWorldBase.cpp#L117

Is this by design?

Note I am working in Unity, but a quick look seems to confirm the same holds for the Unreal implementation.

jonyMarino, 2020-11-13:
Hi @ykeuter! Are you sure that the method is called for the same object twice?

ykeuter, 2020-11-13:
Hi @jonyMarino ! Yes, in the Unity version I stepped through a "tick" in the debugger and can confirm these render methods are called twice: first in `SimModeWorldBase::Tick`, which afterwards calls `SimModeBase::Tick`, which calls` PawnSimApi::pawnTick`, which calls the same render methods again.

I am less familiar with the Unreal implementation code so cannot confirm a similar call stack there at the moment.