microsoft/AirSim #3130: Problem with GPS home location

Agm2104, 2020-11-06:
Hi everyone!

I'm having a problem when trying to run the hello_drone script, since I'm getting the following error:

msgpackrpc.error.RPCError: rpclib: function 'armDisarm' (called with 2 arg(s)) threw an exception. The exception contained this information: Vehicle does not have a valid GPS home location.

However I think I have implemented the location in the settings.json correctly

[image]

Also, when linking with PX4 I get the following:

[image]

Which according with #2647 is a good sign.

However I'm unable to arm the robot and I'm running out of ideas...
What do you think the problem is?

rajat2004, 2020-11-06:
The `params` key should be `Parameters` I think, or you can try to set the parameters yourself in the PX4 terminal

Agm2104, 2020-11-07:
Hey, thanks for answering!
Changing "params" to parameters doesn't seem to work, nor setting the parameters in the PX4 terminal.
One thing I did't mention is that I don't get the 'home set' message on the UE screen, so maybe is not passing the location properly?

jonyMarino, 2020-11-13:
Hi @Agm2104! Which is the player start's location?

Agm2104, 2020-11-13:
Hi @jonyMarino!

These seem to be the start parameters:
x=5690
y=-100
z=202

[image]

ahmed-elsaharti, 2020-11-16:
@Agm2104 , what version of the px4 firmware are you using here? 
Also, do the log messages on the UE environment state that the vehicle is connected to SITL? 
[image]
I don't see anything in the PX4 terminal that confirms that there's a successful connection to the current UE session.
[image]

I tried replicating the error you're getting using your settings.json but everything seemed to work fine.

Agm2104, 2020-11-16:
@ahmed-elsaharti, sorry I didn't add a screenshot of the UE enviroment, I totally forgot. Here it is.

[image]

I'm using the 1.11 version of PX4.

As you say, a confirmation message doesn't appear in the PX4 terminal, but given that i can launch missions with QGC and it's reflected in both UE and PX4, I think they are connected. The problem seems to appear whenever I try to arm the drone via Python scripts, where I get the error mentioned in my first post.

ahmed-elsaharti, 2020-11-16:
@Agm2104, I was able to replicate your results when I switched over to PX4 1.11.2. I can also confirm that with this version the `params` key does not work while `Parameters` does. Try switching over to the version recommended in the docs (1.10.1) and see if the issue persists?

Also, at the player location x=5690 y=-100, the vehicle's GPS coordinates will not be those set by OriginGeopoint. The vehicle's GPS position would be offset from OriginGeopoint by 5690 and -100 meters since OriginGeopoint sets the location of the world origin ( see #2588 )

Agm2104, 2020-11-16:
@ahmed-elsaharti hi,a nd thank you very much for your help!

The problem is that everytime I try to build PX4 with the v1.10.1 I get this error.

[image]

Any ideas about how to fix it?
I'll let you know if the version works the moment it's running :)

ahmed-elsaharti, 2020-11-16:
@Agm2104 No worries.
Which version of the PX4 Windows Toolchain are you using? I used to get all kinds of compiling errors when trying to use 0.9 with PX4 1.10.1 (There was an issue about it where someone pointed out that there was a compiler compatibility issue but I can't find it). Try installing 0.8 and building 1.10.1.
Let us know if it works

Agm2104, 2020-11-16:
@ahmed-elsaharti it's exactly as you say, I was using 0.9 build with 1.11. 
I have changed the versions as you suggested and it works **way** better now, it seems like I really hit the nail with the compatibility.

These are the screens I get now.

[image]
[image]

The only thing that still bugs me is that I doesn't seem to get a **home set** message in the PX4 console, but I'm not entirely sure if that's a problem or not.

Again, thank you very much for your help.

ahmed-elsaharti, 2020-11-16:
@Amg2104 Glad it worked!
As for the **home set** message, If you give it a couple of seconds and try to run `pos` in Droneshell (as is suggested in https://microsoft.github.io/AirSim/px4_sitl/) it should report the GPS coordinates its at rather than 0,0,0 (if you issue `pos` without waiting it will actually report 0,0,0).
What I've seen is that the moment the console reports 'EKF commencing GPS fusion' the position reported by the API (in Droneshell) switches over from 0 to the correct position:
[image]
Not sure if it's directly related to 'GPS fusion' or not, but you can actually see it performing like expected in https://microsoft.github.io/AirSim/px4_sitl/ without having to report **home set**

dayjaby, 2021-03-15:
Seems to be an issue on AirSim side. PX4 confirms that the home position is valid:
```
pxh> listener vehicle_status_flags
...
	condition_home_position_valid: True
```

And still AirSim throws the error:
```
function 'armDisarm' (called with 2 arg(s)) threw an exception. The exception contained this information: Vehicle does not have a valid GPS home location.
```

trungvm, 2021-04-20:
Because of the drone broken before run simulation so you  can make the drone can't be broken in the unreal simulator for fix this problem. It in Player start config. Hope this may help you bro.

jonyMarino, 2021-06-01:
Try with the [latest version](https://github.com/microsoft/AirSim/releases), please

lovettchris, 2021-06-04:
Right, the latest bits should help especially if you also use the latest settings from [here](https://microsoft.github.io/AirSim/px4_sitl/).  Notice I added this block:
```json
            "Sensors":{
                "Barometer":{
                    "SensorType": 1,
                    "Enabled": true,
                    "PressureFactorSigma": 0.0001825
                }
            },
```
which causes the GPS lock to happen a lot more quickly.  The problem because was that PX4 was confused by the random variations in pressure being generated by AirSim barometric pressure simulation.  These new sensor settings clamp down on that randomness which makes PX4 a lot happier.

lovettchris, 2021-07-08:
Closing due to inactivity, please reopen if you are still having this problem with the latest AirSim bits.

closed 2021-07-08 by lovettchris