microsoft/AirSim #3131: fatal error: 'common/AirSimSettings.hpp' file not found

sugawaradaisuke, 2020-11-07:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------
My environments is ubuntu 20.04(NVIDIA Corporation GP107GL [Quadro P620] (rev a1)).
Unreal Engine is 4.24.3.
I built `airsim` and `Unreal Engine` according to [here](https://microsoft.github.io/AirSim/build_linux/).
Following error occur when I open `Blocks` on Unreal Engine.

`[In file included from /home/d/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:2:
In file included from /home/d/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirBlueprintLib.cpp:4:
/home/d/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirBlueprintLib.h:24:10: fatal error: '~/AirSim/AirLib/include/common/AirSimSettings.hpp' file not found
#include "~/AirSim/AirLib/include/common/AirSimSettings.hpp"
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/d/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.gen.cpp:2:
In file included from /home/d/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/AirSim/AirBlueprintLib.gen.cpp:8:
/home/d/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirBlueprintLib.h:24:10: fatal error: '~/AirSim/AirLib/include/common/AirSimSettings.hpp' file not found
#include "~/AirSim/AirLib/include/common/AirSimSettings.hpp"
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
1 error generated.
LogInit: Warning: Still incompatible or missing module: Blocks
LogInit: Warning: Still incompatible or missing module: AirSim](url)`

I can find `AirSimSettings.hpp` at `(AirSim Path)/AirLib/include/common`. How do I fix it ?

rajat2004, 2020-11-09:
There isn't any need to copy the Plugins directory when using the Blocks environment
Try running clean_rebuild.sh, and then do the steps again, just using the scripts and see if that works

jonyMarino, 2021-01-12:
Hi @sugawaradaisuke! And Welcome! Rajat answer is correct so let us know if you still have any issue.

closed 2021-01-12 by jonyMarino