microsoft/AirSim #3132: Solution to error on using build.cmd on Windows

Yaseen787, 2020-11-08:
As I was following the steps to setup AirSim as indicated [here](https://microsoft.github.io/AirSim/build_windows/), I came across the error: `expand-archive : The term 'expand-archive' is not recognized as the name of a cmdlet, function, script file, or operable program.` Apparently, the issue was that the version of PowerShell on my version of Windows was outdated. So, after some searching for a solution I did the following to make things work: I added `set powershell="C:\Program Files\PowerShell\7\pwsh.exe"` at the start of the script and changed all lines that had an occurrence of `Expand-Archive` to have `%powershell%` instead of `powershell `and things seem to have worked.

jonyMarino, 2020-11-13:
Thanks for sharing this! Why did you not update PowerShell?

Yaseen787, 2020-11-14:
Before seeing this error, I had no idea what PowerShell even was, and updating it didn't seem really straight forward. I came to this solution and thought I'd share it here.

closed 2021-03-22 by jonyMarino