microsoft/AirSim #3136: build error

alter409, 2020-11-13:
when I run build.cmd, the error log as follows,how to fix it?

3>已完成生成项目“D:\github\AirSim\Examples\Examples.vcxproj”(默认目标)的操作。
     5>D:\github\AirSim\AirLib\include\vehicles\multirotor\RotorParams.hpp(55,59): error C2220: 以下警告被视为错误 [D:\github\Ai
       rSim\AirLibUnitTests\AirLibUnitTests.vcxproj]
     5>D:\github\AirSim\AirLib\include\vehicles\multirotor\RotorParams.hpp(55,59): warning C4244: “=”: 从“double”转 换到“msr
       ::airlib::real_T”，可能丢失数据 [D:\github\AirSim\AirLibUnitTests\AirLibUnitTests.vcxproj]
     5>D:\github\AirSim\AirLib\include\vehicles\multirotor\RotorParams.hpp(56,88): warning C4244: “=”: 从“double”转 换到“msr
       ::airlib::real_T”，可能丢失数据 [D:\github\AirSim\AirLibUnitTests\AirLibUnitTests.vcxproj]
     9>FinalizeBuildStatus:
         正在删除文件“D:\github\AirSim\MavLinkCom\MavLinkTest\temp\MavLinkTest\x64\Debug\MavLinkTest.tlog\unsuccessfulbuild”。
         正在对“D:\github\AirSim\MavLinkCom\MavLinkTest\temp\MavLinkTest\x64\Debug\MavLinkTest.tlog\MavLinkTest.lastbuilds
         tate”执行 Touch 任务。
     9>已完成生成项目“D:\github\AirSim\MavLinkCom\MavLinkTest\MavLinkTest.vcxproj”(默认目标)的操作。
     5>已完成生成项目“D:\github\AirSim\AirLibUnitTests\AirLibUnitTests.vcxproj”(默认目标)的操作 - 失败。
     1>已完成生成项目“D:\github\AirSim\AirSim.sln”(默认目标)的操作 - 失败。

生成失败。

       “D:\github\AirSim\AirSim.sln”(默认目标) (1) ->
       “D:\github\AirSim\AirLibUnitTests\AirLibUnitTests.vcxproj”(默认目标) (5) ->
       (ClCompile 目标) ->
         D:\github\AirSim\AirLib\include\vehicles\multirotor\RotorParams.hpp(55,59): warning C4244: “=”: 从“double” 转换到“m
       sr::airlib::real_T”，可能丢失数据 [D:\github\AirSim\AirLibUnitTests\AirLibUnitTests.vcxproj]
         D:\github\AirSim\AirLib\include\vehicles\multirotor\RotorParams.hpp(56,88): warning C4244: “=”: 从“double” 转换到“m
       sr::airlib::real_T”，可能丢失数据 [D:\github\AirSim\AirLibUnitTests\AirLibUnitTests.vcxproj]

       “D:\github\AirSim\AirSim.sln”(默认目标) (1) ->
       “D:\github\AirSim\AirLibUnitTests\AirLibUnitTests.vcxproj”(默认目标) (5) ->
       (ClCompile 目标) ->
         D:\github\AirSim\AirLib\include\vehicles\multirotor\RotorParams.hpp(55,59): error C2220: 以下警告被视为错误 [D:\github\
       AirSim\AirLibUnitTests\AirLibUnitTests.vcxproj]

    2 个警告
    1 个错误

已用时间 00:00:19.85

#### Build failed - see messages above.

xxEoD2242, 2020-11-13:
I just had this issue. This is talking about a conversion warning that I'm pretty sure isn't going to be fixed anytime soon, since `msr::airlib::real_T` is a struct with whatever precision they gave it.

What I would recommend (what I did), was go to the file `AirSim\AirLibUnitTests\AirLibUnitTests.vcxproj` and scroll down in the XML until you find a section that looks like this.
[image]

Once you have that, look at the tag that says `DisableSpecificWarnings` and add in this particular warning, which is 4244. So that line for me since I'm using that particular configuration of "Debug | x64", would be:
`<DisableSpecificWarnings>4100;4244;4505;4820;4464;4514;4710;4571;5205%(DisableSpecificWarnings)</DisableSpecificWarnings>`.

Once you have done that, do build.cmd again and it will compile no problem. Since it's only a conversion warning, there won't be any huge issues. Let me know if that works for you!

keliyang, 2020-11-13:
> I just had this issue. This is talking about a conversion warning that I'm pretty sure isn't going to be fixed anytime soon, since `msr::airlib::real_T` is a struct with whatever precision they gave [...]

perfect,I solved my problem,thank you!

alter409, 2020-11-14:
> I just had this issue. This is talking about a conversion warning that I'm pretty sure isn't going to be fixed anytime soon, since `msr::airlib::real_T` is a struct with whatever precision they gave [...]

alter409, 2020-11-14:
> I just had this issue. This is talking about a conversion warning that I'm pretty sure isn't going to be fixed anytime soon, since `msr::airlib::real_T` is a struct with whatever precision they gave [...]

it works for me, thank you!

closed 2020-11-14 by alter409

jonyMarino, 2020-11-16:
Another way would be to add a static cast in the source files. So a PR can be submitted

Alexzzdfjcn, 2020-11-28:
> I just had this issue. This is talking about a conversion warning that I'm pretty sure isn't going to be fixed anytime soon, since `msr::airlib::real_T` is a struct with whatever precision they gave [...]

> I just had this issue. This is talking about a conversion warning that I'm pretty sure isn't going to be fixed anytime soon, since `msr::airlib::real_T` is a struct with whatever precision they gave [...]

that's very helpful! thanks!!

Alexzzdfjcn, 2020-11-28:
> I just had this issue. This is talking about a conversion warning that I'm pretty sure isn't going to be fixed anytime soon, since `msr::airlib::real_T` is a struct with whatever precision they gave [...]

it's very helpful!! thank you!

Rachel-2000, 2021-05-27:
> I just had this issue. This is talking about a conversion warning that I'm pretty sure isn't going to be fixed anytime soon, since `msr::airlib::real_T` is a struct with whatever precision they gave [...]

Works well for my problem, thank you!