microsoft/AirSim #3137: Issue running QGC missions

Agm2104, 2020-11-13:
I'm trying to flight a drone and get LIDAR data by using PX4 with Airsim as SITL. I try to launch a mission in QGC without the sensors enabled and I have no problem, but the moment I enable LIDAR I get 2 issues:

1. When I try to launch my 'default' mission after opening QGC, the drone seems to arm but never takes off and the mission ends.

2. When try to load/upload a different mission, I get the 'mission transfer failed' issue.

I have checked the loss rates (seems to be 0) and the messages on the console logging and I get the following:

[image]

Any idea of what the problem could be?
Here are screenshots of the settings.json file, which seems to be the problem, but I don't see how those parameters affect QGC

Works::
[image]

Doesn't work:
[image]

Any input would be very welcome :)
Thanks in advance!

ahmed-elsaharti, 2020-11-18:
Hi @Agm2104 ,
I ran a couple of tests and was able to replicate your error while using the settings.json that has the sensors by:
1. Launching PX4 SITL
2. Launching the UE4 environment
3. Launching QGC
4. Uploading the mission file which resulted in `mission transfer failed`

The error seemed to go away when launching the simulation in this order:
1. Launch QGC
2. Start PX4 SITL
3. Star the UE4 environment.
4. Upload the mission file and start mission:
[image]
[image]

Hope this helps.

Agm2104, 2020-11-21:
Hey @ahmed-elsaharti, sorry for late answer! Yeah, apparently the error came from a compatibility issue, which was resolved in #3130. Everything seems to run properly now.

Thank you very much for the help!

closed 2020-11-21 by Agm2104