microsoft/AirSim #3138: No x, y horizontal movement API in drone-axis.

ICYPOLE, 2020-11-14:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------
OS: Win10

rlalpha, 2020-11-14:
same problem here too.
Try to connect my remote controller: which have format rc: x_velocity, y_velocity, z_velocity, rotate_velocity

ahmed-elsaharti, 2020-11-18:
You could create a function that uses the pitch, roll and yaw of the vehicle (from `simGetVehiclePose`) to break a velocity vector in the Drone's x,y,z axes to components in the world x,y,z axes and then use `moveByVelocityAsync` to execute those components which would result in an overall motion equivalent to the drone vector you wanted.

Here's a quick 2D example to show what I mean
[image]

jonyMarino, 2020-11-24:
Hey @ICYPOLE! That would be very handy. Want to submit a PR for a new `moveByLocalVelocityAsync` and `setLocalVelocityControllerGainsLocal` with Ahmed's approach?

jonyMarino, 2020-12-01:
Thanks, @ahmed-elsaharti, for your PR! Just for curiosity, what tool are you using to make your graphics?

ahmed-elsaharti, 2020-12-01:
@jonyMarino I generally use [Microsoft Onenote](https://www.onenote.com/) along with my [graphics tablet ](https://www.huion.com/pen_tablet/Huion/NEW_1060PLUS.html) 😃

closed 2020-12-02 by zimmy87