microsoft/AirSim #3141: Any fixed wind drones on AirSim?

Yaseen787, 2020-11-14:
I'm working on a flight control related project and I am exploring several simulators now. As it seems on AirSim, there is no default fixed-wing aircraft as there is for the multirotor already. Is there a source maybe to download more airplane models to use or is it gonna be just the multirotor? I'm not really prepared to go through "hacky" solutions as the project is limited in time. If someone could recommend another simulator which is more suitable I'm more than open.

GimpelZhang, 2020-11-16:
If your project does not have high demand on visual message quality, a gazebo based simulator is a better choice I think. 

For example, the project from ETHZ: [rotors_simulator](https://github.com/ethz-asl/rotors_simulator) has fixed-wing models.

jonyMarino, 2020-11-16:
Hi @Yaseen787! Maybe the work discussed at https://github.com/microsoft/AirSim/issues/2955 in its current development is suitable for you.

AOS55, 2020-11-18:
Hi @Yaseen787, yes if you need visuals for the purpose of debugging and user imagery gazebo may be the way to go, I've been working on adding a native aircraft to airsim but the FDM needs some debugging and the ardupilot interface is a bit of a pain to say the least. I have been working on a python airsim tool to interface but this is really more for RL type stuff. The aim is to integrate them back together once it has the required functionality.

Yaseen787, 2020-11-25:
Hey @AOS55 , thank you for your response. Sounds like a respectful piece of work! How mature would you say the work is? The thing is I need to use something mostly functional for a machine learning project I'm working on, would you say your project would be enough for this? If there is something I can contribute to making the work useable what would it be? Do you know of any other ways of adding fixed-wing models to AirSim?

AOS55, 2020-11-25:
Hi, if its just FCS your interested in I would checkout [Gordon Rennie's](https://github.com/Gor-Ren/gym-jsbsim) Gym environment work you should be up and running quite fast with that its for an RL gym environment and can get visuals with flightgear if you like. What type of machine learning are you planning to use/what is the project?

AOS55, 2020-11-25:
Feel free to use [my python repo](https://github.com/AOS55/Fixedwing-Airsim) too if you like Gordon's doesn't use airsim but has more back tests and is better documented.

Yaseen787, 2020-11-25:
I'm working on a project for autolanding using computer vision techniques, so good visuals are important. The current option I think is most suitable is to use XPlane with Nasa's XPlaneConnect plugin. What do you think?

AOS55, 2020-11-25:
Cool sounds similar to my PhD. Cool XPlane could work well really depends what you want to get out of it. The flight dynamics model is ok it uses an elemental build up approach but no real flight or wind tunnel model data so its quite limited away from a non-linear flight regime but good for basic straight and level descending etc. Visuals are ok, Im not sure how XPlaneConnect works if you go down a ML approach managing data is quite important I'm also not sure if you will need to use an HPC or how to use XPlane on a cluster.

It really depends what you want out of it, if your looking just at the approach mode my python code will give you something to start on with airsim. I have been looking at training a Convnet for semantic segmentation possibly DeepLabV3 or something like that which is doable with how it is at the moment. As Im using RL to control the field selection and approach though reward function is key for me and this comes from collisions which the python solution can't do hence working on the C++ side. If you already know the landing site some CV type approach planning would certainly be interesting and you could use ML for the optimization side of things there is plenty out there on optimal path planning and using ML to replace this. Is this for a university project out of interest?

Yaseen787, 2020-11-30:
Hey, sorry for the late reply, was a bit busy lately. This is for a research project for my master's program, but not the thesis itself, sort of a preparation for the thesis, so I just call it mini-thesis for short. Anyway, I think I'll go for XPlane. AirSim seems to be pro and used more for academics, but I just feel it's too much for me at this point. Just running the default blocks map took all my computer's power, things seem more complicated and there is no default fixed-wing content. This made me sort of shy away from using it. Whereas XPlane seems to run very fine on my computer and running a MATLAB script beside it to access some in-game info (speed, control surface deflections, and so on) and changing the camera view programmatically made me much more comfortable to use it.

My idea right now is to use some MATLAB scripts to spawn the plane at different positions where a plane is likely to be on the final moments of approach, not necessarily on the centerline, and take screenshots programmatically of that, and do a lot of that. Since I can already get the runway start point from the apt.dat file in the XPlane directory, and the aircraft's position could be obtained using the plugin, then I could create a database of runway pictures with the corresponding labels being the angle between the runway and the aircraft direction for example. I could later train a neural network to try and estimate this angle. At this point, I'll have done what's expected of me for the project, but I'm planning to plug this into a path planner, then into a flight controller to have a visual-based auto-landing algorithm.

Since this all will be done on my laptop, I don't think I'll be using cluster computing or any fancy distributed computing thing. The institute I'm doing this with expect me to show some good results before they give me any more resources. I would love to hear what you think of all of this, and any suggestions you may have =)