microsoft/AirSim #3142: Storage capacity and OS of drones in AirSim simulator

AIKUUM, 2020-11-16:
What is the storage capacity of drones in AirSim?
What is the OS supported by drones in AirSim?

jonyMarino, 2020-11-24:
Hey @AIKUUM! I like to see how much you are diving into the project. There is no OS running in the simple-flight drones, but you have a ROS wrapper if that is what you are looking for.
To what are you referring as the 'storage capacity of drones'? How many drones can you instantiate? Not sure now.

AIKUUM, 2020-11-25:
> Hey @AIKUUM! I like to see how much you are diving into the project. There is no OS running in the simple-flight drones, but you have a ROS wrapper if that is what you are looking for.
> To what are you referring as the 'storage capacity of drones'? How many drones can you instantiate? Not sure now.

Assume that a desktop has a storage capacity of 1 TB. Assume that a 32 GB pendrive has a storage capacity of 32 GB.
Though AirSim is a simulator, there must be some storage capacity of a drone right. Something like assume that a drone in AirSim simulator can store a maximum of 1000 images with a resolution of 1000*800. Thats what i mean by "storage capacity of drones.". Please help me with some more information about this if possible.

jonyMarino, 2020-11-25:
There is no simulated disk for drones in AirSim. What do you want to accomplish?

xxEoD2242, 2020-11-25:
@AIKUUM  I think you misunderstand how AirSim and Unreal interact with each other. The virtual drone that is created in Unreal Engine is a component of the memory of the game thread of Unreal Engine. If you look through the code, you will see that each version of the drone is essentially a small block of memory in the expansive memory block that is Unreal Engine.

I think the confusion comes from your notion of real drones, where they run an OS and have some form of memory.

I have been working on something very similar to what you are thinking about. In essence, what I do is simulate the idea of the drone having "memory" by carving out memory in Python that I run on a separate process. The separate process is important, as this will not reduce the memory from Unreal that is needed to run the simulation.

I have a class structure built up for the drone to model exactly what you are talking about. I can send you a small example file if you'd like to see how we think about this but in essence, if you want to simulate the drone having some form of memory, just create the idea of a memory block for the drone and then do everything in Python.

We have thought about adding this into the AirSim as an option but I have to do a lot more research on how Plugins interact with Unreal and how the memory structure would handle that.

AIKUUM, 2020-11-26:
> There is no simulated disk for drones in AirSim. What do you want to accomplish?

@jonyMarino, is there a way to write a program which would consider virtual OS, virtual storage capacity and virtual RAM for a drone in AirSim? Can you send me a link for any such information if its available?

What I would like to accomplish as of now is, I want to communicate between 2 drones in a simulation environment of AirSim. Initially Drone 1 will capture a set of images. I want to transfer those images to Drone 2. Then I should be able to login to Drone 2, get an info that Drone 2 with so and so IP address has received n number of images of so and so sizes from Drone 1 with so and so IP address.

AIKUUM, 2020-11-26:
> @AIKUUM I think you misunderstand how AirSim and Unreal interact with each other. The virtual drone that is created in Unreal Engine is a component of the memory of the game thread of Unreal Engine. [...]

@xxEoD2242, Can you send me the example file, or the link for the same about which you are speaking?

AIKUUM, 2020-12-01:
> @xxEoD2242, Can you send me the example file, or the link for the same about which you are speaking?

@xxEoD2242, Can you send me the example file, or the link for the same about which you are speaking?

AIKUUM, 2020-12-01:
> @jonyMarino, is there a way to write a program which would consider virtual OS, virtual storage capacity and virtual RAM for a drone in AirSim? Can you send me a link for any such information if its [...]

@jonyMarino, Can you give me a hint on how to proceed? Would appreciate it.

xxEoD2242, 2020-12-01:
> @xxEoD2242, Can you send me the example file, or the link for the same about which you are speaking?

I will need a bit of time to refactor the code slightly, as it is part of a wider project that I am working on. However, based off of what you were saying, I would recommend something like the following:

```
import numpy as np

class Drone():

    def __init__(self, ip_address='127.0.0.1'):
        self.ip_address = ip_address
        self.max_storage_size = 1000000 # kb i.e. 1 GB
        self.max_numb_images = 0
        self.specify_memory_size
        self.memory = np.zeros((self.max_numb_images))

    def specify_memory_size(self):
        img_size = 240 * 240 # Number of pixels
        format = 'jpeg'
        # The expected size can be calculated by img_size and format if you look up the compression ratios
        expected_size = 620 # kilobytes
        self.max_numb_images = np.floor(self.max_storage_size / expected_size)

```

I would recommend something similar to this. Then, during execution, store the images in the memory array. You can also get expected size by doing `sys.getsizeof(image)` after capturing the image. When adding to the array, check the length of the array or use numpy ndarray, which won't let you add more items then you have allocated "memory" for. In my very crude example that needs refinement, you would then be able to load the array as much as you want and you could even set up a queue to send images between drones by instantiating n drones and then building TCP-ish packets to send the image to the IP address of the drone.

In this way, you are simulating what would occur with the drone. The project that I am working on will be to do this at the AirSim level, which requires a very large amount of research as I will be running the "memory" component of the drones in a separate process to ensure I don't overtax the game thread of Unreal Engine. As we make progress, I'll share with you the file in this repository: [Multi-Agent Routing Example](https://github.com/xxEoD2242/multi_agent_routing.git) Make sure to keep checking that and we will eventually offer up some of our work as add-ons for AirSim.

AIKUUM, 2020-12-02:
> > > > @AIKUUM I think you misunderstand how AirSim and Unreal interact with each other. The virtual drone that is created in Unreal Engine is a component of the memory of the game thread of Unreal Engine. If you look through the code, you will see that each version of the drone is essentially a small block of memory in the expansive memory block that is Unreal Engine.
> > > > I think the confusion comes from your notion of real drones, where they run an OS and have some form of memory.
> > > > I have been working on something very similar to what you are thinking about. In essence, what I do is simulate the idea of the drone having "memory" by carving out memory in Python that I run on a separate process. The separate process is important, as this will not reduce the memory from Unreal that is needed to run the simulation.
> > > > I have a class structure built up for the drone to model exactly what you are talking about. I can send you a small example file if you'd like to see how we think about this but in essence, if you want to simulate the drone having some form of memory, just create the idea of a memory block for the drone and then do everything in Python.
> > > > We have thought about adding this into the AirSim as an option but I have to do a lot more research on how Plugins interact with Unreal and how the memory structure would handle that.
> > > 
> > > 
> > > @xxEoD2242, Can you send me the example file, or the link for the same about which you are speaking?
> > 
> > 
> > @xxEoD2242, Can you send me the example file, or the link for the same about which you are speaking?
> 
> I will need a bit of time to refactor the code slightly, as it is part of a wider project that I am working on. However, based off of what you were saying, I would recommend something like the following:
> 
> ```
> import numpy as np
> 
> 
> class Drone():
> 
>     def __init__(self, ip_address='127.0.0.1', max_numb_images=0):
>         self.ip_address = ip_address
>         self.max_storage_size = 1000000 # kb i.e. 1 GB
>         self.max_numb_images = max_numb_images
>         if max_numb_images == 0:
>             self.specify_memory_size
>         self.memory = np.zeros((self.max_numb_images))
> 
>     def specify_memory_size(self):
>         img_size = 240 * 240 # Number of pixels
>         format = 'jpeg'
>         # The expected size can be calculated by img_size and format if you look up the compression ratios
>         expected_size = 620 # kilobytes
>         self.max_numb_images = np.floor(self.max_storage_size / expected_size)
> ```
> 
> I would recommend something similar to this. Then, during execution, store the images in the memory array. You can also get expected size by doing `sys.getsizeof(image)` after capturing the image. When adding to the array, check the length of the array or use numpy ndarray, which won't let you add more items then you have allocated "memory" for. In my very crude example that needs refinement, you would then be able to load the array as much as you want and you could even set up a queue to send images between drones by instantiating n drones and then building TCP-ish packets to send the image to the IP address of the drone.
> 
> To move to the next level, you can run each drone as a separate thread of a main process, allowing you to use mutex's to share messages between the threads. You could even run multiple processes and then have a publish-subscribe messaging system built up to share messages between drones. All of this would take a significant amount of work but may replicate what you are seeking.
> 
> In this way, you are simulating what would occur with the drone. The project that I am working on will be to do this at the AirSim level, which requires a very large amount of research as I will be running the "memory" component of the drones in a separate process to ensure I don't overtax the game thread of Unreal Engine. As we make progress, I'll share with you the file in this repository: [Multi-Agent Routing Example](https://github.com/xxEoD2242/multi_agent_routing.git). Make sure to keep checking that and we will eventually offer up some of our work as add-ons for AirSim.
> 
> A more painful but efficient process would be to do all of this in C++, as you can control the memory schema much more finitely. Our eventual work will move in that direction.

Thank you