microsoft/AirSim #3143: Drone does not move in UnityDemo and AirSim on Linux/Ubuntu 20.04 but works with same setup in Windows

CodeQuixote, 2020-11-16:
## Contextual Background
My goal is to simulate a drone using AirSim built on Unity and control it using ROS-MAVROS-PX4 in Linux/Ubuntu 20.04.
I have successfully managed to achieve the above using UnrealEngine, with and without PX4 and ROS with the Python and C++ APIs provided.
I now need to achieve the above with AirSim built on Unity.
After trying with the whole ROS-MAVROS-PX4_AirSim-Unity pipeline, the drone connected but was only able to move its rotors, not the body of the drone. So I decided to start with the simplest setup suggested in the documentation to discard sources of error.

## Setup
* Followed the AirSim on Unity manual: [https://microsoft.github.io/AirSim/Unity/]( https://microsoft.github.io/AirSim/Unity/ ) and cloned the AirSim GitHub repo.
* Built the UnityDemo project on Unity 2019.3.12

## Test
### In Windows:
* I run the HelloDrone AirSim.sln from VisualStudio and everything works.

### In Linux Ubuntu 20.04: 
* I run ./HelloDrone from AirSim output binaries.
* The drone connects as expected.
* I arm the drone and, as in Windows, I get:

`requestApiControl was successful 30 Vehicle=SimpleFlight`
`Vehicle is already armed 30 Vehicle=SimpleFlight`

* I press Enter to takeoff and the rotors start to rotate faster (nb: the drone starts the simulation hovering in the air at a slow rotor speed), but **the drone body does not move** and I don't get the following log that I did observe in Windows:

`API call was not received, entering hover mode for safety 30 Vehicle=SimpleFlight`

* I press Enter to follow the HelloDrone script trajectory and again, in Linux the drone does not move as it does in Windows.
* I then press Enter to first land and then disarm the drone, and now both in Windows and in Linux I get the expected logs that follow:

`Vehicle is disarmed 30 Vehicle=SimpleFlight`
`State:	 Disarmed 30 Vehicle=SimpleFlight`

## Question
* What causes this difference in behaviour for the same, simple, standard project between Windows and Linux/Ubuntu 20.04?
*  How would I fix it or work around it?

Let me know if I have missed anything obvious or if you have any ideas.
Any suggestions will be very much welcome, thanks in advance.

y-elena4, 2022-10-28:
I have the same issue in Linux, did you ever find a solution @CodeQuixote ?

sandilyasg, 2023-11-06:
@y-elena4 @CodeQuixote were you able to find a solution?