microsoft/AirSim #3145: IP addressing in multiple drones, and drone communication protocols

AIKUUM, 2020-11-17:
My system specifications:-
[image]

AirSim Version:- 1.3.1

Unreal Engine Version:- 4.25

Consider a scenario in which I want to transfer a high resolution image and a text file from Drone 1 to Drone 2. Can any 1 tell me what is the best communication protocol I can use for such a communication? 

Also I would like to know with what IP addresses I can configure the 2 drones. Can any one please help

jonyMarino, 2020-11-24:
Hi! Take a look at Mavlink/Mavros. I never tried to communicate something custom from the drone up to now.

AIKUUM, 2020-12-05:
> Hi! Take a look at Mavlink/Mavros. I never tried to communicate something custom from the drone up to now.

Hi @jonyMarino, can Mavlink/Mavros be used in AirSim simulator??

jonyMarino, 2020-12-08:
Hi! Yes, sure. Look for Mavros + AirSim. MavLink is used internally to connect to PX4, so I recommend that you continue with MavROS for your use case (if you can). I would love to see you submit a PR on documentation related to Mavros. Do you think you could?

AIKUUM, 2020-12-09:
> Hi! Yes, sure. Look for Mavros + AirSim. MavLink is used internally to connect to PX4, so I recommend that you continue with MavROS for your use case (if you can). I would love to see you submit a PR on documentation related to Mavros. Do you think you could?

AIKUUM, 2020-12-10:
> Hi! Yes, sure. Look for Mavros + AirSim. MavLink is used internally to connect to PX4, so I recommend that you continue with MavROS for your use case (if you can). I would love to see you submit a PR on documentation related to Mavros. Do you think you could?

Hi @jonyMarino, I will rely on AirSim and one or more related softwares. By any chance is any hardware required for simulating UAV communication in AirSim?

jonyMarino, 2020-12-10:
Not for communications. But from what I see from your first post, I think you are going to have to upgrade your computer. More RAM (at least 32GB). Do you have a GPU?

AIKUUM, 2020-12-11:
> Not for communications. But from what I see from your first post, I think you are going to have to upgrade your computer. More RAM (at least 32GB). Do you have a GPU?

Below is the screenshot of details of my system graphics:-
[image]

Can I work on my system with 8 GB RAM. Or should I compulsorily use my system with 32 GB RAM?

AIKUUM, 2020-12-11:
> Hi! Yes, sure. Look for Mavros + AirSim. MavLink is used internally to connect to PX4, so I recommend that you continue with MavROS for your use case (if you can). I would love to see you submit a PR on documentation related to Mavros. Do you think you could?

HI @jonyMarino, I just want to clarify a small doubt, though I know the fact most probably. Since "SimpleFlight" can be used in AirSim, PX4 need not be used with AirSim right?

jonyMarino, 2020-12-11:
[PX4 ](https://github.com/PX4/PX4-Autopilot) is a separate project, yes. And it can be used together with AirSim.

AIKUUM, 2020-12-14:
> Hi! Yes, sure. Look for Mavros + AirSim. MavLink is used internally to connect to PX4, so I recommend that you continue with MavROS for your use case (if you can). I would love to see you submit a PR on documentation related to Mavros. Do you think you could?

Hi @jonyMarino, Can you please send me some good link for Mavros python package? So that from that I can use the attributes of that package and type my piece of code for communication of data between 2 drones. 
I have downloaded mavros package from the link https://github.com/mavlink/mavros. Should I use some functions from this package?

jonyMarino, 2020-12-14:
I have no material at the moment. Try to get from those projects. Good luck!

AIKUUM, 2020-12-15:
> I have no material at the moment. Try to get from those projects.

Hi @jonyMarino, Can you tell me what extra configurations i should make in settings.json file for communication between 2 UAVs? Should the 2 UAVs be configured with 2 different IP addresses?

jonyMarino, 2020-12-15:
Hi! Try to simulate this on the client-side, using the ROS wrapper. So AirSim is not going to know anything about the communication between drones. I don't have much more to tell you.

AIKUUM, 2020-12-16:
> Hi! Try to simulate this on the client-side, using the ROS wrapper. So AirSim is not going to know anything about the communication between drones. I don't have much more to tell you.

Hi @jonyMarino!! I think you misunderstood my previous comment. I asked if you know about just what configurations should be made like IP addresses for 2 UAVs, port numbers etc in settings.json. Do u have any idea about such configurations in settings.json file?

jonyMarino, 2020-12-16:
AIKUUM. I want to tell you that by having to solve it on the client-side (that is, in your case, the python script that you do for this). All IPs and ports will be in the script, not in seetins.json

AIKUUM, 2020-12-17:
> AIKUUM. I want to tell you that by having to solve it on the client-side (that is, in your case, the python script you do for this). All IPs and ports will be in the script, not in seetins.json
> You must understand that Airsim is a simulator of a drone's physics and sensors. It does not simulate all the details and capabilities of a real drone. Such things like that you have to develop "from the outside".
> It would be nice if you start to see the code inside 💪.

closed 2020-12-22 by AIKUUM