microsoft/AirSim #3148: Airsim fails to compile with VS2019 Community (version 16.8.1)

btwolfe, 2020-11-17:
Running build.cmd in the latest top-level AirSim directory gets fatal errors.
Windows 10, latest AirSim v1.3.1

`D:\AirSim>build.cmd
Found cmake version: 3.18.4
Starting cmake to build rpclib...
-- Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.19041.
-- The C compiler identification is MSVC 19.28.29333.0
-- The CXX compiler identification is MSVC 19.28.29333.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: D:/AirSim/external/rpclib/rpclib-2.2.1/build
Microsoft (R) Build Engine version 16.8.1+bd2ea1e3c for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  Checking Build System
  Building Custom Rule D:/AirSim/external/rpclib/rpclib-2.2.1/CMakeLists.txt
  dispatcher.cc
  server.cc
D:\AirSim\external\rpclib\rpclib-2.2.1\dependencies\include\asio/buffer.hpp(530,5): warning C4834: discarding return va
lue of function with 'nodiscard' attribute [D:\AirSim\external\rpclib\rpclib-2.2.1\build\rpc.vcxproj]
D:\AirSim\external\rpclib\rpclib-2.2.1\dependencies\include\asio/buffer.hpp(529): message : while compiling class templ
ate member function 'void clmdep_asio::detail::buffer_debug_check<std::_Vector_iterator<std::_Vector_val<std::_Simple_t
ypes<_Ty>>>>::operator ()(void)' [D:\AirSim\external\rpclib\rpclib-2.2.1\build\rpc.vcxproj]
          with
          [
              _Ty=clmdep_asio::detail::buffered_stream_storage::byte_type
          ]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\include\type_traits(1584): mess
age : see reference to function template instantiation 'void clmdep_asio::detail::buffer_debug_check<std::_Vector_itera
tor<std::_Vector_val<std::_Simple_types<_Ty>>>>::operator ()(void)' being compiled [D:\AirSim\external\rpclib\rpclib-2.
2.1\build\rpc.vcxproj]
          with
          [
              _Ty=clmdep_asio::detail::buffered_stream_storage::byte_type
          ]
D:\AirSim\external\rpclib\rpclib-2.2.1\dependencies\include\asio/buffer.hpp(1118): message : see reference to class tem
plate instantiation 'clmdep_asio::detail::buffer_debug_check<std::_Vector_iterator<std::_Vector_val<std::_Simple_types<
_Ty>>>>' being compiled [D:\AirSim\external\rpclib\rpclib-2.2.1\build\rpc.vcxproj]
          with
          [
              _Ty=clmdep_asio::detail::buffered_stream_storage::byte_type
          ]
D:\AirSim\external\rpclib\rpclib-2.2.1\dependencies\include\asio/detail/buffered_stream_storage.hpp(57): message : see
reference to function template instantiation 'clmdep_asio::mutable_buffers_1 clmdep_asio::buffer<clmdep_asio::detail::b
uffered_stream_storage::byte_type,std::allocator<clmdep_asio::detail::buffered_stream_storage::byte_type>>(std::vector<
clmdep_asio::detail::buffered_stream_storage::byte_type,std::allocator<clmdep_asio::detail::buffered_stream_storage::by
te_type>> &)' being compiled [D:\AirSim\external\rpclib\rpclib-2.2.1\build\rpc.vcxproj]
  client.cc
D:\AirSim\external\rpclib\rpclib-2.2.1\dependencies\include\asio/buffer.hpp(530,5): warning C4834: discarding return va
lue of function with 'nodiscard' attribute [D:\AirSim\external\rpclib\rpclib-2.2.1\build\rpc.vcxproj]
D:\AirSim\external\rpclib\rpclib-2.2.1\dependencies\include\asio/buffer.hpp(529): message : while compiling class templ
ate member function 'void clmdep_asio::detail::buffer_debug_check<std::_Vector_iterator<std::_Vector_val<std::_Simple_t
ypes<_Ty>>>>::operator ()(void)' [D:\AirSim\external\rpclib\rpclib-2.2.1\build\rpc.vcxproj]
          with
          [
              _Ty=clmdep_asio::detail::buffered_stream_storage::byte_type
          ]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\include\type_traits(1584): mess
age : see reference to function template instantiation 'void clmdep_asio::detail::buffer_debug_check<std::_Vector_itera
[... 1828 lines cut ...]
  LINK : warning LNK4075: ignoring '/INCREMENTAL' due to '/LTCG' specification [D:\AirSim\Unity\AirLibWrapper\AirsimWra
pper\AirsimWrapper.vcxproj]

"D:\AirSim\Unity\AirLibWrapper\AirsimWrapper.sln" (Clean;Build target) (1) ->
"D:\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj.metaproj" (default target) (2:2) ->
"D:\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj" (default target) (5:2) ->
(Link target) ->
  LINK : fatal error LNK1181: cannot open input file 'MavLinkCom.lib' [D:\AirSim\Unity\AirLibWrapper\AirsimWrapper\Airs
imWrapper.vcxproj]

    5 Warning(s)
    1 Error(s)

Time Elapsed 00:02:23.36

D:\AirSim\Unity>if ERRORLEVEL 1 goto :buildfailed

D:\AirSim\Unity>echo(

D:\AirSim\Unity>echo #### Build failed - see messages above.  1>&2
#### Build failed - see messages above.

D:\AirSim\Unity>exit /b 1`

I believe these errors prevent the build process from creating a viable Unity project in the AirSim/Unity/UnityDemo directory.

ahmed-elsaharti, 2020-11-17:
This is probably similar to #3136. Try the workaround explained there or #3146

btwolfe, 2020-11-18:
Well now, that was a painless fix. Allowed the Unity directory build to complete too. Thank you sir. 

Oddly though, the UnityDemo/README.md file talks about opening the UnityDemo project to test the simulation. No such project exists. Fortunately, you can just add the contents of the Assets folder to any project to pull all these details in.

ahmed-elsaharti, 2020-11-18:
@btwolfe Glad that worked!

I haven't tested Airsim with Unity yet, but did you try following https://github.com/microsoft/AirSim/blob/master/docs/Unity.md#usage to run the UnityDemo project?

btwolfe, 2020-11-19:
I reread those instructions and realized that it wasn't suggesting to open the Unity project, just the directory. I didn't know Unity would create a project from the conents it found in a directory. So, cool, that works. Again, thanks for your help.

closed 2020-11-19 by btwolfe