microsoft/AirSim #3149: Attribute

SSZ1, 2020-11-18:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------

SSZ1, 2020-11-18:
I recorded the images in the airsim environment but I can't find out what the content of airsim_rec.txt means, so what does the information in the airsim_rec.txt mean? And I have questions about the  resolution of the images recorded.I edited the width and Height in the camerafaults of Settings.json but that didn't work. I'm confused. Here is the Settings.json I edited.

`{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "Recording": {
    "RecordOnMove": true,
    "RecordInterval": 0.5,
    "Cameras": [
     {
        "CameraName": "0",
        "ImageType": 0,
        "PixelsAsFloat": false,
        "Compress": true
      }
    ]
  },
  "Camerafaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 2560,
        "Height": 1440,
        "FOV_Degrees": 90,
        "AutoExposureSpeed": 100,
        "AutoExposureBias": 0,
        "AutoExposureMaxBrightness": 0.64,
        "AutoExposureMinBrightness": 0.03,
        "MotionBlurAmount": 0,
        "TargetGamma": 1.0,
        "ProjectionMode": "",
        "OrthoWidth": 5.12
      }
    ]
  }
}`

jonyMarino, 2020-11-25:
Hi @SSZ1, and welcome! In airsim_rec.txt, you have information about the vehicle state when the image was taken.
And I can't see from simple sight if something is wrong in your JSON. can you compare with PythonClient\multirotor\high_res_camera.py?

SSZ1, 2022-02-15:
> Hi @SSZ1, and welcome! In airsim_rec.txt, you have information about the vehicle state when the image was taken. And I can't see from simple sight if something is wrong in your JSON. can you compare with PythonClient\multirotor\high_res_camera.py?

Thanks so much