microsoft/AirSim #3150: Ubuntu: How can get global map in Airsim

Han-Sin, 2020-11-18:
oh guys. I do drone simulation in Airsim using unreal Engine. I want to get the global map information from Airsim, but i don't know how to do it. Is there any API for me to do it? Please help me, I really need it.

GimpelZhang, 2020-11-18:
Related issue I guess: #3108 

As the comment mentioned in the above issue, simGetObjectPose and simSetObjectPose can manipulate with objects position ground truth. So I think these may help with the space map generation.

jonyMarino, 2020-11-25:
Hi @HAn-Sin, and welcome!
In this file are the related methods to your question, in particular the ones I have selected.
https://github.com/microsoft/AirSim/blob/659a7dbb93269fba8af31debf6f09f15c3fbc15c/AirLib/include/api/RpcLibClientBase.hpp#L51-L56

Good luck!

jonyMarino, 2021-01-04:
Closed due to inactivity. Please, feel free to ask for it to be reopened if you wish to continue posting.

closed 2021-01-04 by jonyMarino

saihv, 2021-01-04:
Please see https://github.com/microsoft/AirSim/pull/3209 for a way to generate 'global maps' as voxel grids.