microsoft/AirSim #3166: change camera angle

EdwinMeriaux, 2020-11-22:
Hello!

in the standard drone used in Airsim. Is it possible to change the direction the camera is facing?
For example could they be put on the right hand side of the drone facing out or even the bottom of the drone facing down?

GimpelZhang, 2020-11-23:
Hi, 

you can change the direction of camera by editing the setting file. Here is a link to the doc: [Settings](https://microsoft.github.io/AirSim/settings/)

slucasm, 2020-11-23:
Yes @EdwinMeriaux, you have to edit the configuration of "Gimbal" on settings.json.

For facing down the camera:

`"Gimbal": {
        "Stabilization": 1,
        "Pitch": -90, "Roll": 0, "Yaw": 0
      }`

This works for me!

EdwinMeriaux, 2020-11-24:
Awesome! sweet!
Does this mean we could add multiple sets of cameras?
Or are we capped at 1 camera set per drone?

GimpelZhang, 2020-11-24:
> Awesome! sweet!
> Does this mean we could add multiple sets of cameras?
> Or are we capped at 1 camera set per drone?

Yes we can add multiple cameras on one drone. Under "Cameras" section in the setting file, more than one camera can be added and the "X" "Y" "Z" "Pitch" "Roll" "Yaw" attributes control the camera pose. 

Here is my example of two cameras on one car: [example](https://github.com/GimpelZhang/airsim_ros/blob/master/settings/settings.json)

rajat2004, 2020-11-24:
Might be useful - https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/high_res_camera.py

The main part of the PR which added this example got reverted, but example is still present

EdwinMeriaux, 2020-11-25:
Awesome Thank you everyone!

closed 2020-11-25 by EdwinMeriaux