microsoft/AirSim #3167: PX4 + AirSim + QGC

eggbread, 2020-11-24:
Hello,
I am trying to set up the AirSim with multi drones + PX4 sitl + QGC.
I followed up the instruction here

ahmed-elsaharti, 2020-11-24:
Hi @eggbread, from the screenshot, it seems like you're executing `make px4_sitl_default none` rather than `make px4_sitl_default none_iris`. Although this did not make a difference when I tested it (since it tends to default to the iris model anyway), giving it a try won't harm.

Also, which version of the Cygwin Toolchain are you using?

eggbread, 2020-11-25:
Hello @ahmed-elsaharti,
Now,  I fixed it. I had downgrade Cygwin 0.9 to 0.8.Then v1.10.1 is work! Thank you!
From now, I have to connect multiple drones and QGC. 
I am using this settings.
```json
{
    "SettingsVersion": 1.2,
    "SimMode": "Multirotor",
    "Vehicles": {
        "Drone1": {
            "VehicleType": "PX4Multirotor",
            "UseSerial": false,
            "UseTcp": true,
            "TcpPort": 4560,
            "ControlPort": 14580,
            "VehicleSysID": 10,
            "Parameters": {
                "NAV_RCL_ACT": 0,
                "NAV_DLL_ACT": 0,
                "LPE_LAT": 47.641468,
                "LPE_LON": -122.140165,
                "COM_OBL_ACT": 1
            }
        },
        "Drone2": {
            "VehicleType": "PX4Multirotor",
            "UseSerial": false,
            "UseTcp": true,
            "TcpPort": 4560,
            "ControlPort": 14580,
            "VehicleSysID": 20,
            "Parameters": {
                "NAV_RCL_ACT": 0,
                "NAV_DLL_ACT": 0,
                "LPE_LAT": 47.641468,
                "LPE_LON": -122.140165,
                "COM_OBL_ACT": 1
            }
        }
    }
}
```
But the only one drone can be connected with QGC. I saw your suggestion #1599 
In order to do this, What should I do? 
I can't find any video or material about that.
`make px4_sitl_default none_iri` Is it can cover multiple drones?

ahmed-elsaharti, 2020-11-25:
@eggbread sorry for the late reply.
Glad switching over to 0.8 fixed things. I think this will need to be included in the docs since they do mention that PX4 1.10.1 is recommended while it mainly works with Cgwin 0.8.

I tried to look into using multiple drones with SITL. The main issue I found was that `px4_sitl_default` runs one instance of the px4 firmware and I'm assuming we'd need 1 instance per drone. Running two separate instances using Cgwin caused the first instance to shut down (which is probably due to it being made in the same directory). And even if we manage to run two instances using two different installations of the PX4 firmware, there would be a port conflict since AFAIK both instances would be listening to the same port and each drone requires its own port.
I tried looking into https://dev.px4.io/master/en/simulation/multi-vehicle-simulation.html but it seems like the methods provided are very case-specific so I wasn't able to make them work.
If I understand this correctly, if you manage to find a way to change the simulator port on the PX4 SITL instance you could theoretically run two separate instances and tell AirSim (using settings.json) to use two different ports for the drones.

Also, I think the settings file you're using would be slightly problematic since its spawning both drones at the same point and makes them both use the same TCP port to connect to PX4 which would cause a conflict (as seen on the on-screen log in AirSim).
I would recommend:
``` 
{
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "Vehicles": {
      "Drone1": {
          "VehicleType": "PX4Multirotor",
          "UseSerial": false,
          "UseTcp": true,
          "TcpPort": 4560,
          "ControlPort": 14580,
          "VehicleSysID": 10,
          "Parameters": {
              "NAV_RCL_ACT": 0,
              "NAV_DLL_ACT": 0,
              "LPE_LAT": 47.641468,
              "LPE_LON": -122.140165,
              "COM_OBL_ACT": 1
          },
          "X": 0, "Y": 1, "Z": 0
      },
      "Drone2": {
          "VehicleType": "PX4Multirotor",
          "UseSerial": false,
          "UseTcp": true,
          "TcpPort": 4561,
          "ControlPort": 14580,
          "VehicleSysID": 20,
          "Parameters": {
              "NAV_RCL_ACT": 0,
              "NAV_DLL_ACT": 0,
              "LPE_LAT": 47.641468,
              "LPE_LON": -122.140165,
              "COM_OBL_ACT": 1
          },
          "X": 0, "Y": -1, "Z": 0
      }
  }
}
```
This would offset the spawn points and use two different ports.

eggbread, 2020-11-27:
@ahmed-elsaharti 
I left the post in the px4 community.
https://discuss.px4.io/t/multiple-drone-in-the-airsim/19785

I tried like what you said. 
So I can connect PX4 and AirSim with different port number.
But I get the other error :(
I think it is not the error from the AirSim. 
But Could you check it?
Thank you!

ahmed-elsaharti, 2020-11-27:
@eggbread I think I found a way to make this work.
1. Open QGroundControl
2. Open Cygwin and `cd` to PX4/Firmware
3. Run `./Tools/sitl_multiple_run.sh 2`
4. WAIT for QGroundControl to confirm that it made the connection and wait a couple of seconds (since we won't be able to see the debugger, this probably is our best bet at timing this right).
5. Launch UE4/Airsim
6. Wait for GPS lock
[image]
[image]
[image]

To kill the instances run `./Tools/sitl_multiple_run.sh 0`
or `./Tools/sitl_multiple_run.sh 2` to restart the instances

The simulation was slightly wonky but I think it should work.

ahmed-elsaharti, 2020-11-27:
Forgot to mention, I had to get rid of the entire 
```
          "Parameters": {
              "NAV_RCL_ACT": 0,
              "NAV_DLL_ACT": 0,
              "LPE_LAT": 47.641468,
              "LPE_LON": -122.140165,
              "COM_OBL_ACT": 1
          },
```
section for this to work

eggbread, 2020-11-27:
@ahmed-elsaharti 
I am so happy to hear that.
I tried it immediately. Actually, I already tried it before.
In my case, The Vehicle 2 couldn't detect the GPS. and The Vehicle 1 is super wonky.
[image]
And this time is also same.
What is the difference between us?
Here is my settings.json
and I did `./sitl_multiple_run.sh 2`
```json
{
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "Vehicles": {
    "Drone1": {
      "VehicleType": "PX4Multirotor",
      "UseSerial": false,
      "UseTcp": true,
      "TcpPort": 4560,
      "ControlPort": 14580,
      "VehicleSysID": 10,
      "X": 0,
      "Y": 1,
      "Z": 0
    },
    "Drone2": {
      "VehicleType": "PX4Multirotor",
      "UseSerial": false,
      "UseTcp": true,
      "TcpPort": 4561,
      "ControlPort": 14580,
      "VehicleSysID": 20,
      "X": 0,
      "Y": -1,
      "Z": 0
    }
  }
}
```
I am really appreciate for you help :)

ahmed-elsaharti, 2020-11-27:
@eggbread here's my settings.json:
```
{
    "SettingsVersion": 1.2,
    "SimMode": "Multirotor",
    "Vehicles": {
        "Drone1": {
            "VehicleType": "PX4Multirotor",
            "UseSerial": false,
            "UseTcp": true,
            "TcpPort": 4560,
            "ControlPort": 14580,
            "X": 0, "Y": 1, "Z": 0
        },
        "Drone2": {
            "VehicleType": "PX4Multirotor",
            "UseSerial": false,
            "UseTcp": true,
            "TcpPort": 4561,
            "ControlPort": 14580,          
            "X": 0, "Y": -1, "Z": 0
        }
    },
    "OriginGeopoint": {
        "Latitude": 47.641468,
        "Longitude": -122.140165,
         "Altitude": 0
        }
  }
```
the only difference I see is `VehicleSysID`

make sure you wait a couple of seconds after QGroundControl confirms the connection before launching the UE4 environment

ahmed-elsaharti, 2020-11-27:
Also,
if you want to be able to see the px4 messages for each of the instances,
open Cygwin -> cd to Firmware -> execute:
`./build/px4_sitl_default/bin/px4 -i 0 /cygdrive/c/PX4/home/PX4/Firmware/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t /cygdrive/c/PX4/home/PX4/Firmware/test_data`

then open another Cygwin terminal -> cd to Firmware -> execute:
`./build/px4_sitl_default/bin/px4 -i 1 /cygdrive/c/PX4/home/PX4/Firmware/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t /cygdrive/c/PX4/home/PX4/Firmware/test_data`

eggbread, 2020-11-27:
@ahmed-elsaharti 
I want to ask one. Can you run this with stable?
Sometime, My QGC can detect both of GPS. But mostly not.
Even if it can detect, it shows like. I cannot control the Vehicle 2 
[image]

Here is my sequence.
1. Open QGC
2. `./sitl_multiple_run.sh 2`
3. wait a couple of seconds.
4. Check the connection between QGC and PX4.
[image]

5. Start the AirSim.
[image]

It shows continue `Accel - fail:TIMEOUT!` `Gyro 0 fail: TIMEOUT!` error msg..
Anyway, I am in unstable :-(

ahmed-elsaharti, 2020-11-27:
@eggbread did you change the settings.json file? I get the same errors but I'm able to upload and execute a mission before the errors cause disturbance in the behavior of the drone.

Try running the `./sitl_multiple_run.sh 2` command in the blocks environment to see if it makes any difference.

eggbread, 2020-11-27:
@ahmed-elsaharti 
I just tried it.
Unfortunately.. It acts exactly same.
[image]
I think I set up the all options like you.. It is weird.
Unreal Engine : v4.25.4
px4 : v1.10.1
Cygwin : v0.8
QGC : v4.0.11
Visual Studio : 2019
This is my settings. I copied your settings.
```json
{
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "Vehicles": {
    "Drone1": {
      "VehicleType": "PX4Multirotor",
      "UseSerial": false,
      "UseTcp": true,
      "TcpPort": 4560,
      "ControlPort": 14580,
      "X": 0,
      "Y": 1,
      "Z": 0
    },
    "Drone2": {
      "VehicleType": "PX4Multirotor",
      "UseSerial": false,
      "UseTcp": true,
      "TcpPort": 4561,
      "ControlPort": 14580,
      "X": 0,
      "Y": -1,
      "Z": 0
    }
  },
  "OriginGeopoint": {
    "Latitude": 47.641468,
    "Longitude": -122.140165,
    "Altitude": 0
  }
}
```

ahmed-elsaharti, 2020-11-27:
I'm not sure why its not obtaining GPS lock in your case. Try giving it more time?

eggbread, 2020-11-28:
Could you give me version what you use?

eggbread, 2020-11-28:
[image]
OMG..
I did exactly same thing on the ubuntu, it was work!!

ahmed-elsaharti, 2020-11-28:
@eggbread sorry for replying late.
The versions I'm using are:
Unreal Engine: 4.24.3
PX4: 1.10.1
Cygwin: 0.8
QGC: 4.0.11
Visual Studio: 2019

Also, glad it worked on Ubuntu. If you want to do more testing with the Windows version, could you try to reclone the PX4 repo into a new folder and try rebuilding to see if it makes any difference if you start from the beginning?

eggbread, 2020-11-29:
@ahmed-elsaharti 
I found the reason. It was caused by Unreal Engine version.
[image]
In the document, It says Unreal Engine version has to be above 4.24. 
So I installed 4.25. It didn't work for multiple drones.
I think our conversation helps to someone who has same issue.
And In the PX4 [documentation](https://dev.px4.io/master/en/simulation/multi-vehicle-simulation.html), There is no any instruction about Multi-Vehicle with AirSIm. I just want to let you know.
I am so appreciate for your help again.

ahmed-elsaharti, 2020-11-29:
@eggbread Good find!
I just did some testing with 4.25 and I'm getting results much better than those I got with 4.24 (basically the errors are gone and the behavior is still a hit-or-miss but is much more stable). 
[image]
So it looks like the version of UE4 is contributing somewhat to this.

Also, I found a mistake in my instructions, instead of `PX4_SIM_MODEL=iris` try `export PX4_SIM_MODEL=iris` for each instance to be run with the debugging feed.

And yes, I did notice that the PX4 documentation does not mention how Airsim could be run in Multi-Vehicle mode so I'm thinking of trying to submit a PR here with some of that info added to the docs in case others miss this thread while searching through the issues on the repo.

And I'm always glad to be of help 😊

eggbread, 2020-12-14:
Hi, @ahmed-elsaharti 
I have a question. I want to get signal quality from each drone, such as RSSI value.
In other words, it will increase when the drone is far away, and decrease when the drone is near.
Does AirSim have any features?

ahmed-elsaharti, 2020-12-16:
Hi again @eggbread , I dont think airsim has anything like that at the moment, but i think you could create your own using the drone location

eggbread, 2020-12-18:
@ahmed-elsaharti
I saw your pull request, That is so cool!
Anyway, I want to ask a question. Sorry :(
Can I use px4 mode and python API together?
Because If I set the `"VehicleType": "PX4multirotor"` , I couldn't use the python API.
But If I change `"VehicleType": "SimpleFlight"`,then It works!
Is it not support px4 and python API together?

ahmed-elsaharti, 2020-12-18:
Hello again @eggbread 
As far as I remember, the APIs should work fine with PX4 SITL. I remember reading about a bug making this not work at some point but I think it has been fixed (or its the reason we use v1.10.1)

I was able to run PX4 SITL and have the takeoff command work fine:
[image]
using these settings
[image]

eggbread, 2020-12-18:
@ahmed-elsaharti 
Hello! I am sorry for late reply.
Actually, I don't know what it is. I have just rebuild the px4 repo. And then I can use the python API!
Thank you so much!

closed 2021-03-22 by jonyMarino