microsoft/AirSim #3170: Vehicle not spawning at PlayerStart

hazevedosa, 2020-11-26:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------

OS: Windows, AirSim version: 1.3.1, Unreal Version: 4.24

I followed the installation tutorial steps to get the Blocks environment running. However, when I hit "Play" in UE, the vehicle spawns at a completely different position. I can even control the car and hit the obstacles but I can see them -- It looks like all the assets are there, but the visualization on the map is displaced.
[image]

ahmed-elsaharti, 2020-11-26:
Hi @hazevedosa, could you show us your settings.json file? It might shed some light on why this is happening

hazevedosa, 2020-11-26:
I used the minimal settings...:

{
  "SettingsVersion": 1.2,
  "SimMode": "Car"
}

ahmed-elsaharti, 2020-11-26:
@hazevedosa I've done some testing but was not able to replicate your results using the stock start position (i did move the start position up in the Z direction just to make sure it matches the screenshot you posted) and the minimal settings:
[image]
[image]
[image]

I was however able to replicate the start position seen in your screenshots when placing playerstart at 0,0,0:
[image]
and did get some strange behavior when the car is dropped from a relatively high start point.
Have you done any changes to the code and/or the start position? Also, are you using the latest repo?
I suggest going back to the start and recloning the repo in another directory then testing the blocks environment without any changes.

hazevedosa, 2020-11-26:
I am using the latest repo, but I haven't changed the code. 

I can see that when I hit play the PlayerStart Transform immediately changes to (0.0, 0.0, 0.0)... That does not happen when I use AirSim 1.2 in UE 4.18 (I also have those installed).

ahmed-elsaharti, 2020-11-26:
@hazevedosa The transformation to 0,0,0 in the new repo is probably due to #3147. However, that should not change where the player spawns relative to the assets, It only changes the origin of the world to where the player is and re-grabs the player's new location relative to that new origin (which should be 0,0,0 since they're the same point now).

Have you tried redoing the entire process of cloning and building? The reason I'm suggesting redo-ing everything is to make sure this issue is happening because of AirSim and not the clone/build process or anything in between since I'm not able to replicate the results (even though I just recloned and built the latest repo **and** a branch from ~a week ago).

If you do redo everything and end up with the same results, try including what the steps you followed were exactly so that we could try to replicate your results and go from there.

hazevedosa, 2020-11-26:
Ok, @ahmed-elsaharti. I just re-cloned and rebuilt following these steps:

1. Cloned the repo with git clone https://github.com/Microsoft/AirSim.git
2. Got into AirSim folder and ran build.cmd from the command line.
3. Navigated to folder AirSim\Unreal\Environments\Blocks and ran update_from_git.bat.
4. Opened Blocks.sln in Visual Studio 2019 and built with build DebugGame_Editor / Win64 configuration.
5. Pressed the Play button in Unreal Editor. 

Same problem happens -- the car is dropped under the ground.

hazevedosa, 2020-11-26:
So... for now I am using an older version. I downloaded the binaries for 1.3.0 and they work just fine.

ahmed-elsaharti, 2020-11-26:
I realized I was using UE4.25 while trying to replicate this which might make a difference. I'll install 4.24, reclone/rebuild and try to replicate the issue again.
@jonyMarino any ideas?

ahmed-elsaharti, 2020-11-28:
I repeated the test with UE4.24 and was still not able to replicate the problem.

jonyMarino, 2020-11-30:
Hi @hazevedosa! And welcome! When you reported that you were using airsim 1.3.1, you were really using that release or just the latest commit?

hazevedosa, 2020-11-30:
Probably the last commit, as I cloned the repo.

jonyMarino, 2020-12-01:
And what happens if you change the player start position? Does it still spawn at the same place?

hazevedosa, 2020-12-01:
Visually, yes. However, when I change playerStart position, start running and move around, I can realize the car was indeed spawned elsewhere. Just because I can hit different obstacles.

closed 2022-05-30 by jonyMarino