microsoft/AirSim #3171: How to brake UAV during high speed flight ？

love-mj, 2020-11-27:
When I switch to hover mode, the brake will have inertia, and the UAV can't brake immediately. How to achieve this.

ahmed-elsaharti, 2020-11-27:
@love-mj Not sure if I understand the question. But if you want the drone to just instantaneously stop on cue while flying at a high speed, I'm not sure if that's entirely doable just because of the way the simulated physics work (since realistically the drone can't not have inertia).
If that's what you're trying to do, you could always try spawning an object in UE4 right in front of the drone on cue to make it instantaneously collide and stop. I don't think the PX4 autopilot (HITL or SITL) would entirely like that though.

love-mj, 2020-11-27:
@ahmed-elsaharti  What I want to simulate is the hovering of Dajiang obstacle avoidance,I don't know how ？

ahmed-elsaharti, 2020-11-28:
@love-mj I'm not sure I understand what you're trying to do exactly

love-mj, 2020-11-30:
@ahmed-elsaharti What you understand is right. But I have tried many methods, I don't know what function to achieve, how to solve it ？
[image]

lovettchris, 2021-06-04:
If you tell it to fly to the new position away from the wall PX4 will do a pretty aggressive banked turn.  If you also tell it to fly back in the exactly direction it was coming from PX4 will do a steep banked stop to change directions.  You might be able to configure PX4 to be more aggressive up increasing these limits in the [PX4 parameters](https://docs.px4.io/master/en/advanced_config/parameter_reference.html.  Basically check all the MC_*_MAX and the MPC_*_MAX parameters.  I have noticed some of these get reset from time to time, so you can also add these overrides to the AirSim Settings.json file as shown [here](https://microsoft.github.io/AirSim/px4_sitl/).