microsoft/AirSim #3172: Ubuntu how to get the size and shape of the drone?

Han-Sin, 2020-11-27:
I want to know the size and shape of the drone in airsim. And I use "SimpleFlight" in my settings.json.
But I don not find anything about the size in settings.json. So please help me where to find the size of the drone and how to modify it? I really need it, please...

ahmed-elsaharti, 2020-11-28:
@Han-Sin , AFAIK, the dimensions and parameters of the multirotor could be changed within the physics engine using the files in https://github.com/microsoft/AirSim/tree/master/AirLib/include/vehicles/multirotor

For example, the arm length of a generic quad could be set in https://github.com/microsoft/AirSim/blob/659a7dbb93269fba8af31debf6f09f15c3fbc15c/AirLib/include/vehicles/multirotor/MultiRotorParams.hpp#L220 .
As for the visual appearance of the drone, that could be changed by changing the drone's pawn in UE4.

Han-Sin, 2020-12-02:
>  drone's pawn
Thank you very much, but I still can't find where to change the drone's appearance in UE4.
Could you help me more? I really need it. Thank you.

Han-Sin, 2020-12-02:
> @Han-Sin , AFAIK, the dimensions and parameters of the multirotor could be changed within the physics engine using the files in [...]

very strange.When I changed the scale of the drone in ue4, the blades did not turn. I don't know the reason.

qwasdw, 2021-03-20:
Hi, bro, have you solve the problem? The drone's size is 1m x 1m in my UE4 env, i don't know how to change it.