microsoft/AirSim #3174: getLidarData Error After Installing CUDA 10.2.2

dion26, 2020-11-30:
Hallo,

I have an issue with getLidarData function. I didn't change my code and just installed CUDA as mentioned in the title.
It suddenly thrown an error 

  File ".\Lidar_Detection.py", line 565, in <module>
    observe_lidar()
  File ".\Lidar_Detection.py", line 419, in observe_lidar
    pcd, lidar_position, lidar_orientation = generate_geometry(client)
  File ".\Lidar_Detection.py", line 182, in generate_geometry
    lidar_data = client.getLidarData()
  File "C:\Users\diond\Anaconda3\lib\site-packages\airsim\client.py", line 206, in getLidarData
    return LidarData.from_msgpack(self.client.call('getLidarData', lidar_name, vehicle_name))
  File "C:\Users\diond\Anaconda3\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "C:\Users\diond\Anaconda3\lib\site-packages\msgpackrpc\future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: function 'getLidarData' (called with 2 arg(s)) threw an exception. The exception contained this information: No lidar with name  exist on vehicle.

I am not sure why. As I said it did work before and the other function is also still working (getCarState, etc), so it is not a connection problem. Could it be because I just installed CUDA and it updated the NVIDIA Driver, then now it can't read the render result properly?

Thanks for the help.

rajat2004, 2020-12-02:
Are you sure that the Lidar existed in the settings.json? Could you post the settings.json if possible
Might be worth trying out the latest version of Unity as well, that might work with the latest Cuda versions if that's really the cause of the problem

dion26, 2020-12-11:
I am not working on the simulation itself. I only got the project (folder) as is and use it for further use. It worked fine before. I don't know why, suddenly i can't detect the lidar, after installing CUDA and tensorflow. Just uninstalled CUDA the problem presist. I didn't change anything inside simulation folder.

I am trying to run it on the latest airsim and unity now. Any idea how to find more logs, so I can see the problem in more detail?

dion26, 2020-12-11:
So I didn't find the setting.json file but there is only c# script describing the lidar fuction inside the project folder.  As I mentioned before, I don't really know where the setting.json file is. I just received the folder and it ran perfectly in unity.

jonyMarino, 2020-12-11:
Hi @dion26! It is rear to me that you could use lidar if it weren't configured in the seetings.json file.
Can you change the settings to be like described here?
https://microsoft.github.io/AirSim/lidar/

dion26, 2020-12-18:
After I changed the settings.json file to:

`{
 "SimMode" : "", 
 "SettingsVersion" : 1.2, 
 "SeeDocsAt" : "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
 "Lidar2": { 
             "SensorType": 6,
             "Enabled" : true,}
}
`

the Unity crash with log:

`AirsimWrapper.dll caused an Access Violation (0xc0000005)
  in module AirsimWrapper.dll at 0033:c694fac3.`

After searching for a while, it is because i don't have access as an administrator. But why that happened? I just add (enabled) Lidar in the setting file.

RealBrandonChen, 2021-09-04:
For anyone who meets this issue `AirsimWrapper.dll caused an Access Violation (0xc0000005) in module AirsimWrapper.dll at 0033:c694fac3.`  and get an instant crash on Unity activating, I solve this problem by deleting the settings.json and Unity automatically generates the default settings file. Because I had modified the settings.json before to test PX4 connection and the AirsimWrapper may have issue initializing with the settings.

jonyMarino, 2022-02-03:
Unity doesn't have a Lidar implementation, which is the cause of these issues.

closed 2022-02-03 by jonyMarino