microsoft/AirSim #3176: rpc::timeout problem

lknnnli, 2020-12-01:
Hello everyone:
I try to catch images from airsim drone with ros. but when i run client.simGetImages(request), always there would be rpc::timeout:timeout while calling RPC function "simGetImages".

Is there anyone who faces the same problem with me?
Sincerely thank you for your help.

skykim0609, 2021-02-10:
I get the same issue when I simulate the drone with px4 in AirSim.

lovettchris, 2021-06-04:
what happens if you use simGetImage instead as shown in the opencv_show.py script?

sandilyasg, 2024-01-22:
@lknnnli @skykim0609 any update on this?

skykim0609, 2024-01-23:
> @lknnnli @skykim0609 any update on this?

I have not been using AirSIm since 2021, so sorry to tell you that I have no idea on whether this issue is fixed or not.