microsoft/AirSim #3177: the number of recorded depth image and scene image is not consistent

lucafei, 2020-12-01:
hello, guys, i met a problem when i record the depth and scene images by using record button on simulation window. The problem is that i find the number of depth images is not consistent with the number of recorded scene images. I think that is due to the synchronize issue. And meanwhile if i set the attribute PixelsasFloat for depth image in setting.json as true, then the depth image will not be recorded. That confuses me. Do you guys have any idea to solve this problem? Thank you in  advance.

rajat2004, 2020-12-02:
Could you post the exact settings.json used, that's very useful to reproduce the problem

lucafei, 2020-12-02:
> Could you post the exact settings.json used, that's very useful to reproduce the problem

here is my setting.json:

{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "ViewMode": "",
  "Recording": {
    "RecordOnMove": false,
    "RecordInterval": 0.2,
    "Cameras": [
      {
        "CameraName": "0",
        "ImageType": 0,
        "PixelsAsFloat": false,
        "Compress": true
      },
      {
        "CameraName": "0",
        "ImageType": 5,
        "PixelsAsFloat": false,
        "Compress": true
      },
      {
        "CameraName": "0",
        "ImageType": 2,
        "PixelsAsFloat": true,
	"Compress": false
      }
    ]
  },
  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 1024,
        "Height": 720,
        "FOV_Degrees": 100,
        "AutoExposureSpeed": 100,
        "AutoExposureBias": 0,
        "AutoExposureMaxBrightness": 0.64,
        "AutoExposureMinBrightness": 0.03,
        "MotionBlurAmount": 0,
        "TargetGamma": 1.0,
        "ProjectionMode": "",
        "OrthoWidth": 5.12
      },
      {
        "ImageType": 5,
        "Width": 1024,
        "Height": 720,
        "FOV_Degrees": 100,
        "AutoExposureSpeed": 100,
        "AutoExposureBias": 0,
        "AutoExposureMaxBrightness": 0.64,
        "AutoExposureMinBrightness": 0.03,
        "MotionBlurAmount": 0,
        "TargetGamma": 1.0,
        "ProjectionMode": "",
        "OrthoWidth": 5.12
      },
      {
        "ImageType": 1,
        "Width": 1024,
        "Height": 720,
        "FOV_Degrees": 100,
        "AutoExposureSpeed": 100,
        "AutoExposureBias": 0,
        "AutoExposureMaxBrightness": 0.64,
        "AutoExposureMinBrightness": 0.03,
        "MotionBlurAmount": 0,
        "TargetGamma": 1.0,
        "ProjectionMode": "",
        "OrthoWidth": 5.12
      },
      {
        "ImageType": 2,
        "Width": 1024,
        "Height": 720,
        "FOV_Degrees": 100,
        "AutoExposureSpeed": 100,
        "AutoExposureBias": 0,
        "AutoExposureMaxBrightness": 0.64,
        "AutoExposureMinBrightness": 0.03,
        "MotionBlurAmount": 0,
        "TargetGamma": 1.0,
        "ProjectionMode": "",
        "OrthoWidth": 5.12
      }
    ]
  },
  "SegmentationSettings": {
    "InitMethod": "",
    "MeshNamingMethod": "",
    "OverrideExisting": false
  },
  "Vehicle": {
    "SimpleFlight": {
      "VehicleType": "SimpleFlight",
      "RC": {
        "RemoteControlID": 2
      }
    },
    "Cameras": {
      "front_left": {
        "CaptureSettings": [
          {
            "ImageType": 0,
            "Width": 1024,
            "Height": 720,
            "FOV_Degrees": 100
          }
        ],
        "Pitch": 0,
        "Roll": 0,
        "Yaw": 0
      },
      "front_right": {
        "CaptureSettings": [
          {
            "ImageType": 0,
            "Width": 1024,
            "Height": 720,
            "FOV_Degrees": 100
          }
        ],
        "Pitch": 90,
        "Roll": 0,
        "Yaw": 0
      },
      "bottom_center": {
        "CaptureSettings": [
          {
            "ImageType": 0,
            "Width": 1024,
            "Height": 720,
            "FOV_Degrees": 100
          }
        ],
        "Pitch": 0,
        "Roll": 0,
        "Yaw": 0
      },
      "front_center": {
        "CaptureSettings": [
          {
            "ImageType": 0,
            "Width": 1024,
            "Height": 720,
            "FOV_Degrees": 79.7,
            "AutoExposureSpeed": 100,
            "AutoExposureBias": 0,
            "AutoExposureMaxBrightness": 0.64,
            "AutoExposureMinBrightness": 0.03,
            "MotionBlurAmount": 0,
            "TargetGamma": 1.0
          },
          {
            "ImageType": 5,
            "Width": 1024,
            "Height": 720,
            "FOV_Degrees": 79.7,
            "AutoExposureSpeed": 100,
            "AutoExposureBias": 0,
            "AutoExposureMaxBrightness": 0.64,
            "AutoExposureMinBrightness": 0.03,
            "MotionBlurAmount": 0,
            "TargetGamma": 1.0
          },
          {
            "ImageType": 7,
            "Width": 1024,
            "Height": 720,
            "FOV_Degrees": 79.7,
            "AutoExposureSpeed": 100,
            "AutoExposureBias": 0,
            "AutoExposureMaxBrightness": 0.64,
            "AutoExposureMinBrightness": 0.03,
            "MotionBlurAmount": 0,
            "TargetGamma": 1.0
          }
        ],
        "Pitch": 0,
        "Roll": 0,
        "Yaw": 0
      },
      "back_center": {
        "CaptureSettings": [
          {
            "ImageType": 0,
            "Width": 1024,
            "Height": 720,
            "FOV_Degrees": 100
          }
        ],
        "Pitch": 0,
        "Roll": 0,
        "Yaw": 0
      }
    }
  },
  "SubWindows": [
    {
      "WindowID": 0,
      "ImageType": 5,
      "CameraName": "front_center",
      "Visible": true
    },
    {
      "WindowID": 1,
      "ImageType": 0,
      "CameraName": "front_center",
      "Visible": true
    },
    {
      "WindowID": 2,
      "ImageType": 3,
      "CameraName": "front_center",
      "Visible": true
    }
  ]
}

jonyMarino, 2020-12-08:
Hi @lucafei! Can you also provide AirSim and UE versions, please?

lucafei, 2020-12-08:
> Hi @lucafei! Can you also provide AirSim and UE versions, please?
> And I think I already see this issue. Did you search previously for it?

i use latest version of airsim from master and UE4.25.  
i searched this similar issue, but it is about nonconsistent between depth image and scene image.