microsoft/AirSim #3179: ArduRoverApi.hpp can not be compiled

liang-tang, 2020-12-03:
I want to send brake control to AirSim while ardupilot rover simulating.
and i modified ArduRoverApi.hpp to receive brake message, but seems never received. so made the test below
1.execute clean.cmd to clean first compile.
2.delete ArduRoverApi.hpp.
3.run build.cmd to compile airsim
but the compile sitll can be finished.
someone can tell me how to compile it? thanks

rajat2004, 2020-12-04:
The unmodified file compiles and after making the changes, it's compiling, is that so?
You'll have to post atleast the error messages and mostly the modified code also (in a branch) for someone to help you in figuring out the problems

liang-tang, 2020-12-05:
> The unmodified file compiles and after making the changes, it's compiling, is that so?
> You'll have to post atleast the error messages and mostly the modified code also (in a branch) for someone to help you in figuring out the problems
As you can see, some file already been deleted, but when ran build.cmd no error occur.
Why i did this test, I want to modify the Api of ArduRover.
[image]

liang-tang, 2020-12-05:
> The unmodified file compiles and after making the changes, it's compiling, is that so?
> You'll have to post atleast the error messages and mostly the modified code also (in a branch) for someone to help you in figuring out the problems

OK, now i know the whole development workflow, build.bat will copy some files(i.e ArduRoverApi.hpp) to Plugins AirSim folder, and compile it when compile Blocks project.
Also thanks for your reply. @rajat2004

closed 2020-12-05 by liang-tang