microsoft/AirSim #3181: Linking external dependency to AirSim

AOS55, 2020-12-03:
Hey all, I was hoping to get some advice on linking another project to be used AirSim. Related to #2955 I am looking to use [JSBSim](https://github.com/JSBSim-Team/jsbsim) a flight dynamic model with AirSim to simulate fixedwing aircraft. I am not sure what the recommended way is to link an external dependency. 

In visual studio I can see that in the Unreal solution the Game Blocks project has a deps directory under Plugins/AirSim/Source/AirLib/deps. The JSBSim source I am planning to use has a project file and includes CMake files, they also state CMake [is their official method to build the program](https://jsbsim-team.github.io/jsbsim-reference-manual/mypages/quickstart-building-the-program/). But running this project alone with its own visual studio solution successfully compiles the necessary exe file on my computer, but Im not sure if this needs to be changed to include a library?

Ideally I will interface to the [FGFDMExe](https://github.com/JSBSim-Team/jsbsim/blob/master/src/FGFDMExec.h) class in AirSim then use this to update the pawn and interface via RpcLib setting controls and outputting state to control my aircraft. This is the [interface class](https://github.com/AOS55/AirSim-1/blob/FixedWing-JSBSim/AirLib/include/physics/JSBSimPhysicsEngine.hpp) I have written on my branch that is based on updateable object. 

Are there any recommended methods for including deps libraries with AirSim? I don't want to break the cross-compatability so I'm guessing you need to set up CMake & MSBuild? I might also open an issue with this on JSBSim to see how they recommend including the project as a dep of another project. Thanks for everyones help 😄

AOS55, 2020-12-09:
Update on this issue, I have taken the following steps to get a build working but I am still experience problems with the linker in visual studio, E1696 type errors where C++ cannot open the referenced header file. 

[In travis](https://travis-ci.com/github/AOS55/AirSim-1/builds/207325037) all the builds now pass the build command where I have added JSBSim to the [build.cmd](https://github.com/AOS55/AirSim-1/blob/FixedWing-JSBSim/build.cmd) and [setup.sh](https://github.com/AOS55/AirSim-1/blob/FixedWing-JSBSim/setup.sh) & [build.sh](https://github.com/AOS55/AirSim-1/blob/FixedWing-JSBSim/build.sh) scripts respectively. The only difference with this is that JSBSim is made with an install directory so I externally make and install the program then in cmake I have [a macro that adds a subdirectory](https://github.com/AOS55/AirSim-1/blob/FixedWing-JSBSim/cmake/cmake-modules/CommonSetup.cmake#L12) similar to the way eigen is deployed.

I believe these are now all linking and building correctly given the travis builds all pass. With Visual Studio I have added the relevant lines to the [Airlib.vcxproj file](https://github.com/AOS55/AirSim-1/blob/FixedWing-JSBSim/AirLib/AirLib.vcxproj). In particular added all my [JSBSim airlib class header files](https://github.com/AOS55/AirSim-1/blob/FixedWing-JSBSim/AirLib/AirLib.vcxproj#L130) and [JSBSim airlib source files](https://github.com/AOS55/AirSim-1/blob/FixedWing-JSBSim/AirLib/AirLib.vcxproj#L212). I have also added the [jsbsim include header files](https://github.com/AOS55/AirSim-1/blob/FixedWing-JSBSim/AirLib/AirLib.vcxproj#L297) and [JSBSim static library files](https://github.com/AOS55/AirSim-1/blob/FixedWing-JSBSim/AirLib/AirLib.vcxproj#L318) generated by the JSBSim [build.cmd](https://github.com/AOS55/AirSim-1/blob/FixedWing-JSBSim/build.cmd#L203). I have also gone into visual studio and added ..\..\Plugins\AirSim\Source\Airlib\deps\JSBSim\include to the VC++ include directories for block properties in VS. 

Does anyone know what I am missing here, I think I have gone through all the steps I can think of to link the JSBSim library into AirSim but can't work out what I'm missing with VS?

seanmcleod, 2020-12-10:
> I am still experience problems with the linker in visual studio, E1696 type errors where C++ cannot open the referenced header file

You mention the linker, but Exxx errors are Intellisense errors not compiler or linker errors.

Compiler errors and warnings are Cxxxx and linker errors are LNxxxx.

https://docs.microsoft.com/en-us/cpp/error-messages/compiler-errors-1/c-cpp-build-errors?view=msvc-160

So if you aren't seeing any Cxxxx or LNxxxx errors then everything is compiling and linking fine and if you only see Exxxx type errors then it's just an Intellisense issue.

See the following link on how to troubleshoot Intellisense errors.

https://docs.microsoft.com/en-us/cpp/build/cppproperties-schema-reference?view=msvc-160

AOS55, 2020-12-10:
He Sean thanks for your help and explanation. Changing the #include path I had set it to `<JSBSim/include/FGFDMExec.h>` for the header reference in the airsim/airlib header file. The E1696 code was caused by vs saying that it can't find the header file I changed this to a relative reference in the header file `#include "Airlib/deps/JSBSim/include/JSBSim/FGFDMExec.h"` and now I get the error: 

`Severity	Code	Description	Project	File	Line	Suppression State
Error	C1083	Cannot open include file: 'simgear/compiler.h': No such file or directory	Blocks	C:\Users\quessy\Documents\Unreal Projects\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\Airlib\deps\JSBSim\include\JSBSim\simgear\props\props.hxx	24	
`

[Referring to the micosoft docs](https://docs.microsoft.com/en-us/cpp/error-messages/compiler-errors-1/fatal-error-c1083?f1url=%3FappId%3DDev16IDEF1%26l%3DEN-US%26k%3Dk(C1083)%26rd%3Dtrue&view=msvc-160) the error is because the file is not included on the search path. But I thought this was done by adding the header directory to the VC++ include directories in the properties directory. Shown here, my only other thought is that the librarys included are not in the path
[image].

For the include I took the header files under the include directory generated by `cmake . install` from JSBSim and the librarys from the Debug & Release directorys made by the cmake VS generator. This is done in [build.cmd](https://github.com/AOS55/AirSim-1/blob/FixedWing-JSBSim/build.cmd#L203), in VS the project solution filters then have:
[image]

Do you know if there is something I need to add in VS for JSBSim's include header files to be on the same path? Or is there a search path I need to include for the library apart from what is already within [Airlib.vcxproj file](https://github.com/AOS55/AirSim-1/blob/FixedWing-JSBSim/AirLib/AirLib.vcxproj)?

seanmcleod, 2020-12-10:
If it's only an IntelliSense error why not ignore it for now and confirm that the integration between AirSim and JSBSim works? 

I would've thought you could use `#include <FGFDMExec.h>` in AirLib given you have the include path set for JSBSim.

AOS55, 2020-12-10:
I think its more so if I use `#include <JSBSim/FGFDMExec.h>` the linter states can't find directory 'JSBSim' in search path and the compiler error is returned `Severity	Code	Description	Project	File	Line	Suppression State
Error	C1083	Cannot open include file: 'JSBSim/FGFDMExec.h': No such file or directory	Blocks	C:\Users\quessy\Documents\Unreal Projects\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\physics\JSBSimPhysicsEngine.hpp	9	
` 

So it can't seem to see the header file on the path, with an absolute path then it can't see the header files that are referenced in FGFDMExec.h.

xxEoD2242, 2020-12-10:
@AOS55 Is it just the Blocks plugin that has is throwing the issue with the compiler?

I'll pull your branch and start helping with integration tomorrow after my finals are finished. Would have started helping sooner but school comes first unfortunately.

AOS55, 2020-12-10:
Hi @xxEoD2242 yes thats the one, yes of course nw, thanks for your help.
I just rebuilt the whole thing and it now seems to be happy with the JSBSim/include header directory but the simgear directory is having problems I'm going to try including this on the path with VC++ directories too as I think this needs referencing from the relative path too.

AOS55, 2020-12-10:
Hey the latest commit technically compiles but I had to put in a more relative header in simgear eg in sg_path.hxx I took `#include <simgear/compiler.h>` and added `#include <JSBSim/simgear/compiler.h>` if included in the project correctly this should be fine under VC++ directories Im not sure why its not doing this though.

AOS55, 2020-12-15:
Hey quick update Ive been debugging the FixedWing-JSBSim branch. I did not have the class in the [SimHUD.cpp](https://github.com/AOS55/AirSim-1/blob/FixedWing-JSBSim/Unreal/Plugins/AirSim/Source/SimHUD/SimHUD.cpp#L310) and I did not have jsbsim in the [settings file](https://github.com/AOS55/AirSim-1/blob/FixedWing-JSBSim/AirLib/include/common/AirSimSettings.hpp#L801) which is now included.

Using Visual Studio when running with JSBSim in the settings.json file:
```{
  "SettingsVersion": 1.2,
  "SimMode": "JSBSim",
  "Vehicles": {
    "JSBSimPlane": {
      "VehicleType": "jsbsim"
      }
    },
  "SubWindows": [
    {
      "WindowID": 0,
      "CameraName": "0",
      "ImageType": 3,
      "Visible": true
    },
    {
      "WindowID": 1,
      "CameraName": "1",
      "ImageType": 5,
      "Visible": true
    },
    {
      "WindowID": 2,
      "CameraName": "2",
      "ImageType": 0,
      "Visible": true
    }
  ]
 }
```

The program stops with an exception at line 1701 of the xtree library with exception: `Exception thrown: read access violation.
_Scary->_Myhead was nullptr.`. I believe this is caused by the `connect_member` command [PawnSimApi::initialize()](https://github.com/AOS55/AirSim-1/blob/FixedWing-JSBSim/Unreal/Plugins/AirSim/Source/PawnSimApi.cpp#L53). Does anyone know what this is for? I have tried looking at the command with breakpoints in Multirotor but could not work it out or work out why the error message occurs.

AOS55, 2021-03-29:
Closing this issue problem documented in #2955 link working different problem occuring

closed 2021-03-29 by AOS55