microsoft/AirSim #3182: Is there an autopilot we can use for flying the drone? (Ubuntu 18.04, AirsSim v1.0-linux, Python3)

alik-git, 2020-12-04:
Hi there, I'm working on an ML project and I'm trying to make a deep learning model learn how to navigate a quadrotor though the square obstacles in the Final Tier 2 Level in Airsim. At the moment we wrote a manual hacky controller for it in python that maps keyboard presses to yaw changes and eventually calls the moveByVelocityAsync function, but we'd like to use a PID controller to get it to follow a predetermined path. Is there such a pid controller available that I can use or do I have to write one myself? If so, whats the best way of doing that?

Basically we are trying to use reinforcement learning and have the PID controller correct the drone when it's flying off trajectory, while recording the controls it outputs so we can use the correction as new data to improve our model's navigation ability. 

If you need more information feel free to ask.

My questions are:
Is there a PID controller or an Autopilot function somewhere I can call?
If I have to write one myself, whats the best / cleanest way of doing that?
Are there any pre-trained models online that follow a predetermined trajectory well that I can use as my autopilot?

jonyMarino, 2020-12-11:
Hi @alik-git! And welcome! The drone already has a PID controller that is used when you move with the move* APIs.
You can integrate with PX4 SITL/HITL and QGroundControl if you need a path planner.
For the pre-trained models, we will submit a PR for DQN example that works for the landscape mountains env. Is that what you are looking for? Take a look at the pull requests late next week.

kartikeya13, 2021-03-20:
hey ali,
Can you share that function? 
I m trying to collect image data by moving the drone in the env.

closed 2021-07-23 by jonyMarino