microsoft/AirSim #3183: AirSim upgrading project to Unreal 4.26.0 error

tom-dv, 2020-12-04:
Windows 10, AirSim v1.3.1, Unreal 4.26.0

Testing Blocks project with AirSim plugin in Unreal 4.26.0, build fails in VS2019 with the following errors.

```
Error	LNK2001	unresolved external symbol "__declspec(dllimport) private: static class UClass * __cdecl UPhysicalMaterial::GetPrivateStaticClass(void)" (__imp_?GetPrivateStaticClass@UPhysicalMaterial@@CAPEAVUClass@@XZ)	Blocks	E:\Git\AirSim\Unreal\Environments\Blocks\Intermediate\ProjectFiles\Module.AirSim.cpp.obj	1	

Error	LNK1120	1 unresolved externals	Blocks	E:\Git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Binaries\Win64\UE4Editor-AirSim.dll	1	

Error	MSB3073	The command ""C:\Program Files\Epic Games\UE_4.26\Engine\Build\BatchFiles\Build.bat" BlocksEditor Win64 Development -Project="E:\Git\AirSim\Unreal\Environments\Blocks\Blocks.uproject" -WaitMutex -FromMsBuild" exited with code 6.	Blocks	C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.MakeFile.Targets	46	
```

The Blocks project with AirSim plugin work fine in Unreal 4.25.4.

jairotunior, 2020-12-07:
I have the same problem!!!

botkevin, 2020-12-09:
I have the same problem. Pretty sure there's just an error with the Airsim plugin with the new engine version. Switching to 4.24.3 fixed the issue for me.

misakabribri, 2020-12-10:
Unreal 4.25 is ok. Unreal 4.24.3 is ok (recommanded version). I have the same problem when I try to use the 4.26. Actually, you can install different versions of unreal from Epic.
[image]

nikhil-sethi, 2020-12-30:
Facing the same issue on Ubuntu 20.04 with the following error and some PhysX depreciation warnings.
AirSim v1.3.1, Unreal 4.26.0 (release)
```
[4/6] Link (lld) libUE4Editor-AirSim.so
ld.lld: error: undefined symbol: UPhysicalMaterial::GetPrivateStaticClass()
>>> referenced by PhysicalMaterial.h:27 (Runtime/PhysicsCore/Public/PhysicalMaterials/PhysicalMaterial.h:27)
>>>               <AirSim Path>/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp.o:(ACarPawn::ACarPawn())
>>> referenced by PhysicalMaterial.h:27 (Runtime/PhysicsCore/Public/PhysicalMaterials/PhysicalMaterial.h:27)
>>>               <AirSim Path>/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp.o:(ACarPawn::ACarPawn())
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
```

jonyMarino, 2021-01-06:
Hi @nikhil-sethi! Can you try with this PR #3277, please? And give us some feedback?

nikhil-sethi, 2021-01-07:
Hi @jonyMarino. Thank you for the reply. It doesn't seem to be working unfortunately and throws the exact same error. @zimmy87 Is there anything more I can do to test this? I tried a new clone and rebuild of AirSim. Should I try and rebuild Unreal as well or are there any smaller tests that I can carry out?

jonyMarino, 2021-01-07:
Thanks for your test! Did you check out the PR, or you tested master?

nikhil-sethi, 2021-01-07:
I checked out the PR by fetching changes and creating a new branch. Verified the changes (they were very small in this case) and ran the editor with Blocks.uproject at the command line.

zimmy87, 2021-02-02:
Hi @nikhil-sethi, can you retry the build after deleting the Binaries, Saved, and Intermediate folders in your Unreal environment? Sometimes Unreal needs to completely regenerate all intermediate build files in order to succeed.

nikhil-sethi, 2021-02-10:
Hi @zimmy87. Sorry for the delay, I wasn't available. Just to confirm, how do I retry the Blocks build after deleting the binaries folder? Should I rebuild(`Setup.sh -> GenerateProjectFiles.sh -> make `) the entire Unreal engine release(4.26 branch) from scratch and a git clone?

jonyMarino, 2021-02-10:
Hi @nikhil-sethi! Thanks for continuing to help. Could you try cloning into another folder and compiling AirSim from 0? That way, we would have no doubt.

nikhil-sethi, 2021-02-12:
Hi @jonyMarino. I'm happy to help. I tried a complete rebuild of Unreal(tag:[4.26.0-release]) and AirSim([commit]) as well and its working as expected. I've tried the both builds two times so there shouldn't be any doubt. Thanks for the help!
I'm figuring my way out through the repo for development and would love your feedback on what went wrong or right. Was it #3277 or in general broken files that were fixed on a fresh build?

[4.26.0-release]: https://github.com/EpicGames/UnrealEngine/tree/4.26.0-release
[commit]: https://github.com/microsoft/AirSim/commit/0096b5e7e841dd2534ddb43c4cb03d0c5e491577

jonyMarino, 2021-02-16:
Hi @nikhil-sethi! Good that you want to delve deeper into the repository for development! Count on me!
I think some intermediate file was not being deleted, so I asked you to start with a new clone. The only fix for this was #3277. Good luck!

PR #3277 (2021-01-07): Fix issue with building after upgrading UE 4.26
add extra dependency on "PhysicsCore" to load definition for UPhysicalMaterial::StaticClass symbol

Fixes: #3183 

## About
This PR fixes a LNK2001 error that occurs when building the AimSim plugin with Unreal Engine 4.26. The LNK2001 error occurs due to a missing object for the UPhysicalMaterial::StaticClass symbol. This is happening because the UPhysicalMaterial class was moved from the "Engine" module to the "PhysicsCore" module. This PR updates the build configuration for the AirSim plugin to add an extra dependency for the "PhysicsCore" modu [... cut]
closed 2021-02-16 by jonyMarino