microsoft/AirSim #3184: takeoffheight modification

EdwinMeriaux, 2020-12-05:
So  def takeoffAsync(self, timeout_sec = 20, vehicle_name = ''):

makes the drone take off 3 meter above the starting point.
Can I modify this to 1m or .5 meters?

ahmed-elsaharti, 2020-12-05:
Hi @EdwinMeriaux , i think the takeoff height can be changed [here](https://github.com/microsoft/AirSim/blob/f6553d4d609b774ee8584dc91e6a22185824cf42/AirLib/include/vehicles/multirotor/firmwares/simple_flight/firmware/Params.hpp#L115) and/or [here](https://github.com/microsoft/AirSim/blob/f6553d4d609b774ee8584dc91e6a22185824cf42/AirLib/include/vehicles/multirotor/firmwares/mavlink/MavLinkMultirotorApi.hpp#L465)

EdwinMeriaux, 2020-12-06:
Hi! thanks for the answer. So I want to change the takeoff height to .2 meters so I modified both values to -0.2. But that does not seem to change anything...

rajat2004, 2020-12-06:
@EdwinMeriaux Have you tried using the [`moveToZAsync`](https://github.com/microsoft/AirSim/blob/master/PythonClient/airsim/client.py#L887) API?
Modification of the values not causing any changes seems strange, assuming it's using the updated values and not earlier files, maybe the height is too low for the speed?  The speed seems to be hardcoded to [0.5 m/s](https://github.com/microsoft/AirSim/blob/master/AirLib/src/vehicles/multirotor/api/MultirotorApiBase.cpp#L35)

ahmed-elsaharti, 2020-12-06:
@EdwinMeriaux what flightcontroller mode are you using? is it Simpleflight?
If it is, after you modify [this line](https://github.com/microsoft/AirSim/blob/f6553d4d609b774ee8584dc91e6a22185824cf42/AirLib/include/vehicles/multirotor/firmwares/simple_flight/firmware/Params.hpp#L115) you'll need to rebuild the project using build.cmd for changes to take effect.

If you're using the blocks environment to test this, make sure you rebuild the project within the Blocks solution in VS2019.
[image]

EdwinMeriaux, 2020-12-06:
Hello!

So I tried moveToZAsync and it did not work:
client.moveToZAsync(-1,-0.1).join()

The drone did no take off

So, I am using the default flight controller, is that simpleflight?
I am not using the blocks environment but the Subt environment and from Linux. How do I rebuild?
https://github.com/microsoft/AirSim/releases/tag/v1.2.0Linux

ahmed-elsaharti, 2020-12-06:
@EdwinMeriaux I did some testing and it seems like 0.1-0.2 meters is a bit too small of a value as @rajat2004 mentioned.

Also, AFAIK these are compiled environments so you cant rebuild/change them. If you want to test code changes easily I suggest that you have a look at https://microsoft.github.io/AirSim/unreal_blocks/

EdwinMeriaux, 2020-12-06:
Ok Sweet!

Thank you so much!

I will test in the blocks environment.
But for a general note. Those prebuilt environments, is there a way to get the Unreal Project so they can be eventually rebuilt and changed? I would like to work using a subterranean world and that Subt is perfect!

ahmed-elsaharti, 2020-12-06:
Glad to be of help ^_^

Also, I don't think the Unreal Project would be retrievable from those environments. Especially since some of them are not even free (like the [neighborhood project](https://www.unrealengine.com/marketplace/en-US/product/modular-neighborhood-pack) for example). I could be wrong though.

You could look around in the Unreal marketplace to see if you can grab something that suits your needs (they have free ones too) and then follow [this guide ](https://microsoft.github.io/AirSim/unreal_custenv/) to incorporate Airsim into the environment you download. The guide has multiple links to sites and tutorials that should help you find an environment.

EdwinMeriaux, 2020-12-06:
Awesome!
Thank you so much!

closed 2020-12-06 by EdwinMeriaux