microsoft/AirSim #3186: Rover Control issue

liang-tang, 2020-12-05:
I'm trying to send brake control to rover in Ardurover simulation, but strange for me is even if the brake is very small, the car still there, and the RPM is always below 1450. I tested a few number like 0.00001, -0.00001,  the car never move.

[image]

liang-tang, 2020-12-06:
@rajat2004 Would you have a look at this issue?

rajat2004, 2020-12-25:
Sorry, completely missed this
Quite strange, and to confirm the car does move if the value is 0? I don't remember AirSim doing anything with the brake or even the other values itself, and it was just directly passed to the PhysX interface, but need to check

liang-tang, 2020-12-26:
@rajat2004 
From the results of my test, if the value is not 0(e.g 0.0001), the action is same as when the value is 1. I think brake is just controlled by true(not 0, enable brake) and false(0, disable brake).