microsoft/AirSim #3187: Does there exist a way to circumvent GPS lock

Deepakgthomas, 2020-12-06:
Airsim doesn't allow my vehicle to arm indoors because of the GPS lock issue. However, I would be totally fine without the GPS. Could someone suggest a way to stop Airsim from looking for GPS lock before arming?

jonyMarino, 2020-12-15:
Hi @Deepakgthomas! Can you refer to the "GPS lock issue" you are talking about? is it the one related to PX4?

Deepakgthomas, 2020-12-15:
Hi @jonyMarino! 
Firstly thank you for responding to this. I and my team are running Airsim on a Jetson Nano which is placed on top of our drone. When we try to arm our drone through Airsim, it doesn't let us do that saying it could not get a GPS lock. We would still like to fly our drone, despite the poor GPS signal indoors and therefore looking to turn off the request for GPS lock.
Please let me know if you have more questions.

lovettchris, 2021-06-04:
This is a PX4 question, you could try setting the `COM_ARM_WO_GPS` - see https://docs.px4.io/master/en/advanced_config/parameter_reference.html

Deepakgthomas, 2021-06-04:
Thank you for the information. We paused that project a few months ago and therefore I will close this issue.

closed 2021-06-04 by Deepakgthomas