microsoft/AirSim #3189: MovetopositionAsync not working with drivetrain set to ForwardOnly

Villotadavid, 2020-12-06:
Hello!
I am implementing an algorithm that creates random trajectories to use reinforced learning techniques afterwards.
To ensure that the drone goes forward I was intended to use the "_ForwardOnly_" drivetrain type.

However, when I run the code, the drone remains steady. To check if the process has finished I print the "_set_flag"  variable of 
 "_client.moveToPositionAsync_" (Future object) and I can see that it is always set to "_True_"

I have already had a look around the existing issues and I think this question is similar to #1131 however it was not properly formulated and it does not have a solution.

I am using Airsim 1.3.0 and Python 3.7.6

Here my start class:

    def start(self,trajectory,ax):   #trajectory is an array of points and ax is a figure object to plot the drone´s position
        print("arming the drone...")
        self.client.armDisarm(True)
       
        start = self.client.getMultirotorState().kinematics_estimated.position
        landed = self.client.getMultirotorState().landed_state
        if not self.takeoff and landed == airsim.LandedState.Landed: 
            self.takeoff = True
            self.client.takeoffAsync().join()
            start = self.client.getMultirotorState().kinematics_estimated.position

        else:
            print("already flying")

        start = self.client.getMultirotorState()
        print (start)
        
        print ('Drone en posición de inicio')
        position = self.client.getMultirotorState().kinematics_estimated.position
        print (position)
        n=0
        droneline, =ax.plot3D([0], [0], [0],color='blue',alpha=0.5)
        
        for point in trajectory:
            print (point)
            a=self.client.moveToPositionAsync(int(point[0]), int(point[1]), int(point[2]), 2, drivetrain = 1) #Set drivetrian to ForwardOnly
            data=self.client.getMultirotorState()
            collision=self.client.simGetCollisionInfo()
            while not(a._set_flag) and not(collision.has_collided):  #Check whether it has finished or collided
                data=self.client.getMultirotorState()
                collision=self.client.simGetCollisionInfo()
                position = data.kinematics_estimated.position
                update_line(droneline,[position.x_val ,position.y_val,-position.z_val])   #Update the plot
                plt.pause(0.25)
           

        print("landing...")
        self.client.landAsync().join()

        print("disarming.")
        self.client.armDisarm(False)

ahmed-elsaharti, 2020-12-06:
Hi @Villotadavid, try 
`client.moveToPositionAsync(int(point[0]), int(point[1]), int(point[2]), 2, 3e+38,airsim.DrivetrainType.ForwardOnly, airsim.YawMode(False,0))` 
Let us know if it works

Villotadavid, 2020-12-07:
Hello @ahmed-elsaharti
Thanks for your prompt reply!
It works perfectly. 
I close the issue.
Thank you very much!

closed 2020-12-07 by Villotadavid