microsoft/AirSim #3190: Where to put new Blueprint

EdwinMeriaux, 2020-12-07:
Hi!

So for the past few weeks have been trying to import a custom drone into airsim.
I built the uasset file. But I think I am storing it in the wrong place. 
in: https://github.com/Microsoft/AirSim/issues/838

it says to specify the drone here:
 "DefaultQuadrotor": {"PawnBP":"/AirSim/Blueprints/BP_FlyingPawn"} 
in the Json file. Where is this? I am trying to find it. 
So I found many BP_FlyingPawn but I don't know which is right

in my computer I found:
- /home/useraccount/AirSim/Unreal/Plugins/AirSim/Content/Blueprints/BP_FlyingPawn
- /home/useraccount/AirSim/Unreal/Environments/LandscapeMountains 4.24/Plugins/AirSim/Content/Blueprints/BP_FlyingPawn
- /home/useraccount/AirSim/Unreal/Environments/LandscapeMountains/Plugins/AirSim/Content/Blueprints/BP_FlyingPawn
- /home/useraccount/AirSim/Unreal/Environments/Blocks 4.24/Plugins/AirSim/Content/Blueprints/BP_FlyingPawn
- /home/useraccount/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Blueprints/BP_FlyingPawn
- /home/useraccount/AirSim/Unreal/Environments/Blocks 4.24/Saved/EditorCooked/LinuxNoEditor/Blocks/Plugins/AirSim /Content/Blueprints/BP_FlyingPawn

But I cannot find: /AirSim/Blueprints/BP_FlyingPawn

ahmed-elsaharti, 2020-12-07:
Hi @EdwinMeriaux, I haven't done as much testing with this as I'd like to be able to give useful advice, but I would suggest placing the uasset file under /home/useraccount/AirSim/Unreal/Environments/Blocks 4.24/Content/TestFolder so that you'd be able to view it from the unreal editor:
[image]
[image]
then go about making the changes seen [here ](https://github.com/Microsoft/AirSim/blob/v1.1.10/docs/settings.md#pawnpaths) (ie: make the parent class FlyingPawn, etc..) to make it useable with Airsim. After that, you could point to it in settings.json through 
```
    "DefaultQuadrotor": { "PawnBP": "Class'/Game/TestFolder/mionioassembly.mionioassembly_C'" }
```

I'm not an expert on the topic though, so I might be completely off here.

EdwinMeriaux, 2020-12-07:
Hi! yes any help is greatly appreciated at this point.
So would you need to put the whole folder of the uasset and not just the .uasset file?

ahmed-elsaharti, 2020-12-07:
Not necessarily (as far as i understand). Im just suggesting that you place it in a folder so that its viewable in your unreal editor so you could go in and change the parent class and all

EdwinMeriaux, 2020-12-07:
Ok that makes sense.
One of the major parts that confuses me is how do we change the parent class?
What do we change it to?

ahmed-elsaharti, 2020-12-07:
@EdwinMeriaux as [the guide](https://github.com/Microsoft/AirSim/blob/v1.1.10/docs/settings.md#pawnpaths) suggests, double click on the asset:
[image]
Then click on "Class Settings" in the menu bar of the new window that opens:
[image]
then you'll be able to change the Parent Class:
[image]

I'm guessing you would want to change it to "Flying Pawn" rather than the "Car Pawn" specified in the guide

EdwinMeriaux, 2020-12-08:
Hi!

So to build a new drone would it probably be best to take this existing framework with a the current default drone and swap the Target? Because a new uasset would not have all these prebuilt no? (the default class for an imported asset from solidworks appears to be Level Script Actor.

ahmed-elsaharti, 2020-12-08:
I haven't tried this before but if i were you I'd start with the default uasset like you just said. It'd probably be much easier this way

EdwinMeriaux, 2020-12-08:
Ok that makes sense.
Sorry back tracking a bit.
I opened up the block environment in unreal. Where is the default drone uasset?
It's not in my content browser. Did you import it directly?

ahmed-elsaharti, 2020-12-08:
If i remember correctly, i copied it from /Plugins/AirSim/Content/Blueprints/BP_FlyingPawn and pasted into a new folder under /Blocks 4.24/Content/ Renamed it from within UE then renamed the file itself and pointed to it in setting.json (i renamed it to test the pointing from settings.json)

Again, the way im suggesting probably isnt the best way but is probably the easiest since you'd be modifying an already working uasset file rather than trying to figure out how to make one from scratch

EdwinMeriaux, 2020-12-09:
Ok so now I have my custom drone and the default drone in the blocks environment.
I think the major confusion left is how to set up the custom drone with all the properties of the other drone. Should I just rebuild the class? and the prop states? Since you said building from scratch might be better?

ahmed-elsaharti, 2020-12-09:
If I were to do this I'd modify the default drone and adding my custom mesh to it first to see how things work before starting from scratch. Starting from scratch would probably be the better way but not the easier way.

Especially since the default drone already has the prop states and the cameras included, so I'd start by just replacing the UAV's mesh and seeing how things go.

EdwinMeriaux, 2020-12-19:
Yes I will try to modify the frame of drone.
Do you if I modify the frame of the drone in a manner that would affect the position of the props that would be problematic? I would need to just move the location of the props right?

Sorry for the delay in response on my end. I just finished my finals for the semester and now can get back to work. I hope the delay is not seen as an insult or anything. I just had zero time this last 1.5 weeks.

ahmed-elsaharti, 2020-12-19:
@EdwinMeriaux I think that should be fine. However, those changes will only appear visually. To make them effective in the physics simulation you would have to adjust the changed parameters (length of arm holding the rotor, etc) in [MultiRotorParams](https://github.com/microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/MultiRotorParams.hpp). But again, start changing one thing at a time. So start with the mesh first then once that's taken care of try looking at the simulation parameters.

Also, no worries at all, I'm in the EXACT same boat actually 😂😂😂

EdwinMeriaux, 2020-12-20:
@ahmed-elsaharti 
Yes I do see that the modifications would be only visual but I need to to resemble the right drone.

So I made the easiest modification could to the base drone mesh. I scaled it down to be much smaller. So this new drone lets call it BP_FlyingPawnsmall. It is currently stored in the contents file of my project Blocks4.24. I tried openning it in Airsim by modifying the json file but it is not working right.
The following error keeps showing up when I hit start:
Error at startup: Failed to load asset class - Class'/Blocks 4.24/Content/BP_FlyingPawn.BP_FlyingPawn_C'

Here is an copy of my json file that I was using:

{
	"SettingsVersion": 1.2,
	"SimMode": "Multirotor",
	"Vehicles": {
		"Drone1": {
			"Sensors": {
			    "Barometer": {
			         "SensorType": 1,
			         "Enabled" : true
			    },
			    "Imu": {
			         "SensorType": 2,
			         "Enabled" : true
			    },
			    "Gps": {
			         "SensorType": 3,
			         "Enabled" : true
			    },
			    "Magnetometer": {
			         "SensorType": 4,
			         "Enabled" : true
			    },
			    "Distance": {
			         "SensorType": 5,
			         "Enabled" : true
			    },
			    "Lidar2": {
			         "SensorType": 6,
			         "Enabled" : true,
			         "NumberOfChannels": 4,
			         "PointsPerSecond": 10000
			    }
			},
			"VehicleType": "SimpleFlight",
			"X": 0, "Y": -37, "Z": 0
		}
		
	},
	

  	"PawnPaths": {
    "DefaultQuadrotor": {"PawnBP": "Class'/Blocks 4.24/Content/BP_FlyingPawnsmall.BP_FlyingPawnsmall_C'"},
 
	
	"number": 1,
	"names": ["Drone1"]

}
}

I'm willing to bet the issue is in the location I put for the defaultQuadrotor I tried many modified addresses for the drone but I cannot seem to figure out what is the right one.

the docs say:

"PawnBP":"Class'/Game/MyCar/MySedanBP.MySedanBP_C'" if the drone is stored in: Content\MyCar\MySedanBP.uasset

But that also does not appear to work or at least I'm doing that wrong aswell :)

ahmed-elsaharti, 2020-12-21:
@EdwinMeriaux if your asset is placed in `/Blocks 4.24/Content/` then your settings.json should be pointing to `/Game/`

ie: 
```json
"PawnPaths": {
"DefaultQuadrotor": {"PawnBP": "Class'/Game/BP_FlyingPawnsmall.BP_FlyingPawnsmall_C'"},
```
similar to what the docs say (I'm assuming that your file is `BP_FlyingPawnsmall.uasset`)

Also, make sure the file name is correct, your error message says `BP_FlyingPawn` and the settings you posted say `BP_FlyingPawnsmall` so there might be a file name issue somewhere.

EdwinMeriaux, 2020-12-21:
Awesome! Yes that worked perfectly!
I tried so many different paths to the drone and non worked until this!

So if this drone is current in the blocks 4.24 project. If I have another project I would need to import it into that project. So if I would store it in the airsim plugin folder it could be accessed from all projects rigth?

ahmed-elsaharti, 2020-12-21:
@EdwinMeriaux Not entirely sure tbh. Give it a try though

EdwinMeriaux, 2020-12-24:
Ok so I tried it. But it does not work. I will try to modify the original file for the default drone. I will see if that works.

closed 2021-01-09 by EdwinMeriaux

airsimdevdd, 2021-02-25:
@EdwinMeriaux @ahmed-elsaharti how to import my customized drone which is in the fbx format from unreal engine to airsim environment ,i am a newbie so please tell me the step by step process of this ?

ahmed-elsaharti, 2021-02-26:
@airsimdevdd here's a video that shows how I go about adding a custom drone.
https://drive.google.com/file/d/1-iOfJ8zO3YA4Mzc0ZkKwTM3L4--4RFtt/view?usp=sharing
Note that this only changes the LOOK of the drone and does not affect the physics of the simulation.

airsimdevdd, 2021-02-26:
@ahmed-elsaharti thank you sir how can i contact you personally for further assistance can i get your contact ,I have to complete this project so i request you to assist me

airsimdevdd, 2021-02-26:
@ahmed-elsaharti sir my custom drone is not a quadcopter. it is bit complex one that is co axial octocopter , we dont have a lot of support ,resources &experience . In our case we cant simply modify the default flying pawn we had to change physics like arm count ,motor count and lot  .u already helped us with normal one so  please help us out with main one also ,thanks in advance

ahmed-elsaharti, 2021-02-26:
> @ahmed-elsaharti sir my custom drone is not a quadcopter. it is bit complex one that is co axial octocopter , we dont have a lot of support ,resources &experience . In our case we cant simply modify [...]

@airsimdevdd Not entirely sure how you'd be simulating a coaxial octocopter on AirSim. 
Firstly, I'm pretty sure AirSim does not support the frame you're looking for. So you will have to add it manually into [Multirotorparams](https://github.com/microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/MultiRotorParams.hpp)
You might also need to go and look into the physics sim to see if you need to make more modifications to accommodate for the drone you're trying to simulate
Moreover, (AFAIK) simpleflight doesnt even support hexacopters so I highly doubt that using it here would even be an option. PX4 SITL or HITL might be an option but you would have to do some digging into how to add a custom HITL frame (its somewhat covered [here](https://github.com/microsoft/AirSim/wiki/hexacopter) for a hexarotor). If you decide to go down the SITL route, I'll be submitting a PR soon showing how to add a custom SITL airframe that contains a hexacopter mixer, you might be able to use the steps from that along with a coax octo mixer (if one exists).

airsimdevdd, 2021-02-26:
Thank you for your response.

airsimdevdd, 2021-02-27:
@ahmed-elsaharti sir ,I am facing some small issues regarding import my fbx model into UE4 editor it is importing as different parts I tried making it one peice but failed i need your assistance to overcome  with this problem sir and i have many small doubts for that i need your  personal mail or any personal contact  if u can share to my mail airsimdev@gmail.com
                     thanks in advance...

ahmed-elsaharti, 2021-03-01:
@airsimdevdd I'm not very experienced in the topic but I would suggest using something like Fusion360 or Blender to combine the files. Also, I'd honestly rather try to help out here on the repo so that if someone else runs into a similar issue they'd be able to find our discussion and benefit from it. It's also a better idea to continue the discussion over at #3426 as to not hijack an already closed issue post.

aaqibism, 2021-04-15:
@ahmed-elsaharti I wanted to follow up on how to adjust the SITL airframe. How do we go about changing the airframe when running in none_iris mode. Should it be done within Airsim or in QgroundControl? It seems like QgroundControl only changes the airframe for HITL. Thank you.