microsoft/AirSim #3191: Vehicle does not have a valid GPS home location.

xzh1049090448, 2020-12-07:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------

ahmed-elsaharti, 2020-12-07:
This is probably similar to #3130.

xzh1049090448, 2020-12-08:
ok， thank you

closed 2020-12-08 by xzh1049090448