microsoft/AirSim #3192: unexpected behavior on Multirotor “moveonpath”

lucafei, 2020-12-07:
hi, guys, i'm recently using the api moveonpath to execute a customized path, but i met a unexpected behavior on this function. i have put corresponding codes as following:
   # move to the predefined position
    client.moveToPositionAsync(1000, -10, -60, 10).join()
    # move to the end position
    airsim.wait_key('Press any key to land')
    client.moveOnPathAsync([airsim.Vector3r(100, -10, -30),
                            airsim.Vector3r(50, -5, -15),
                            airsim.Vector3r(10, 5, -10),
                            airsim.Vector3r(0, -2, -5),
                            airsim.Vector3r(-60, 0, -1),
                            airsim.Vector3r(-120, 0, -1)], 5, 120, airsim.DrivetrainType.ForwardOnly,
                           airsim.YawMode(False, 0), 20, 1).join()
the problem is that the multirotor always early stops at the position, which even not through the first point (100, -10,-30).   but if i set the startpoint smaller like as (300,-10,-60), then it will work and move the endpoint(-120,0,-1). I 'm confused fot that. Do you have any ideas for this and how to solve this? Thank you in advance!

ahmed-elsaharti, 2020-12-07:
Hi @lucafei,
The distance between (1000,-10,-60) and (100, -10, -30) is 900.5 meters. At 5m/s (the speed in your command) it would take 180 seconds to complete the first motion. You have the command timeout set to 120 seconds which is why the command stops 2/3 the way through the first motion. Try increasing the timeout.

lucafei, 2020-12-07:
aha, @ahmed-elsaharti  thank you for your advice. i will try it.

lucafei, 2020-12-07:
it works :) i will close this issue

closed 2020-12-07 by lucafei