microsoft/AirSim #3198: Compilation Error - Airsim Block.uproject - Ubuntu 20.04

ferrLucas, 2020-12-08:
Hi, I am having issues when trying to run Blocks.uproject, after a fresh installation following the instructions:

git clone https://github.com/Microsoft/AirSim.git
cd AirSim
./setup.sh
./build.sh
./Engine/Binaries/Linux/UE4Editor
followed by browsing to AirSim/Unreal/Environments/Blocks, and choosing the convert-in-place option (with no extra dialogs until failure).

System details: Ubuntu 20.04 / UE Editor 4.27.0 / Python 3.8.5 / updated Master branch of Airsim 

Problem happens around 14% of the C++ compilation step, giving those errors:

2020.12.08-16.22.56:202][419]/home/lferr/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/Vehicles/Car/CarPawnApi.cpp:63:15: error: member access into incomplete type 'physx::PxVehicleWheels'
[2020.12.08-16.22.56:202][419]            pv->mWheelsDynData.setToRestState();
[2020.12.08-16.22.56:202][419]              ^
[2020.12.08-16.22.56:202][419]/home/lferr/UnrealEngine/Engine/Plugins/Runtime/PhysXVehicles/Source/PhysXVehicles/Public/WheeledVehicleMovementComponent.h:20:8: note: forward declaration of 'physx::PxVehicleWheels'
[2020.12.08-16.22.56:202][419]        class PxVehicleWheels;
[2020.12.08-16.22.56:202][419]              ^
[2020.12.08-16.22.56:202][419]In file included from /home/lferr/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:27:
[2020.12.08-16.22.56:202][419]/home/lferr/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/Vehicles/Car/CarPawnApi.cpp:67:16: error: member access into incomplete type 'physx::PxVehicleDrive'
[2020.12.08-16.22.56:202][419]            pvd->mDriveDynData.setToRestState();
[2020.12.08-16.22.56:202][419]               ^

Which I believe is the cause from the project not being compiled successfully. 
[error.txt](https://github.com/microsoft/AirSim/files/5660491/error.txt)

ahmed-elsaharti, 2020-12-09:
Hi @ferrLucas, have you tried compiling with versions of UE4 known to be compatible with AirSim?
AFAIK the version known to be supported is 4.24. But I've also used 4.25 without problems.

ferrLucas, 2020-12-10:
Hi @ahmed-elsaharti, I just tried now with 4.25.4 and it worked! Thank you.

closed 2020-12-10 by ferrLucas

leekyyoon99, 2024-12-30:
I appreciate your answer too bro!!!