microsoft/AirSim #3200: Blueprint compile error (OptionsMenu) in Unreal 4.25

alonfaraj, 2020-12-09:
Hi,

I'm using Unreal **4.25.4** on Win10.
Seems like `Plugins\AirSim\Content\HUDAssets\OptionsMenu` failed to compile:

[image]

Tested on Blocks environment as well.

petergerten, 2020-12-20:
I have the same issue

saihv, 2020-12-22:
This is really odd. It looks like this is the commit that's causing the #3200 issue. The broken reference is this file that was deleted.

https://github.com/microsoft/AirSim/commit/bb1b7e3d5ec841ae86cebab3f8d5f9ed3e7743a7#diff-2d1bbba2ccc6fc0c43288832ef48c9c3d9efb5d326a6f5a404ac6a4c7875ff84

@clovett Chris, could you let us know what issue you were facing prior to your commit?

saihv, 2020-12-22:
@petergerten @alonfaraj  I quickly threw together a fix in https://github.com/microsoft/AirSim/pull/3240, could you please give this a shot to see if it fixes the compile errors?

lovettchris, 2020-12-23:
Oh, hmmm, interesting, I was getting a fatal compile error when cooking assets of the Neighborhood environment because BP_LevelLoadButton.uasset was unused.  Without this asset I was able to cook the environment (meaning File/Package Project/Windows/Windows (64bit)) with the asset I get this fatal error:
```
UATHelper: Packaging (Windows (64-bit)):   LogBlueprint: Error: [Compiler BP_LevelLoadButton] In use pin  Menu Widget  no longer exists on node  Get . Please refresh node or break links to remove pin. from Source: /AirSim/HUDAssets/BP_LevelLoadButton.BP_LevelLoadButton
```

saihv, 2020-12-24:
Huh, interesting. Looks like the internal links in the options menu were
broken even before this commit then.

I restored the asset and fixed the broken links in
https://github.com/microsoft/AirSim/pull/3240 and it seems like packaging
is working (according to CI).

lovettchris, 2020-12-24:
Ok, if you've tested packaging works, then you should merge your PR then, thanks!

alonfaraj, 2020-12-28:
@saihv Any plans to merge it if packaging works?

closed 2020-12-28 by jonyMarino