microsoft/AirSim #3201: [Unreal Engine Crash unpredictable when training DRL]

changpowei, 2020-12-10:
## Hardware/Software Information
- CPU: Intel® Xeon(R) Silver 4116 CPU @ 2.10GHz × 24
- GPU: NVIDIA Corporation TU102 [TITAN RTX]
- RAM: 64GB
- OS: ubuntu 20.04
- python: 3.8
- aisim version: 1.3.0
- Unreal Engine version: 4.24.3-0+++UE4+Release-4.24
- Environment: AirSimNH

## Error screenshot
-I am using RL to train a drone avoid obstacle in AirSimNH environment. It seems working perfectly over whole day and night(around 50000 steps in 4000 episodes), but it suddenly crash with following error log.

[image]
[image]
[image]

saihv, 2020-12-10:
Looking at the log, it seems like you're running out of GPU memory. I would recommend running `watch -d -n 0.5 nvidia-smi` in another terminal and monitor the GPU memory usage.

changpowei, 2020-12-10:
@saihv Thanks for replying, I will keep watching.

changpowei, 2020-12-10:
Hi @saihv 
I found that the GPU memory usage of AirSimNH is keep increasing. It started from about 1100MB and continues to accumulate to 1600MB so far. What could be the cause of this situation? 

[image]

wangwwno1, 2020-12-14:
Hello, apart from the GPU memory please also check the configuration files used by the environment.
If the format of settings.json is incorrect, it would raise similiar error.
Plus, please check the training code as it occupy over 21GB, periodically dump unused cache(especially for PyTorch) would help.
[image]

changpowei, 2020-12-15:
Hi @wangwwno1 ,

I set the fixed GPU memory size when running the training code, so it occupy the 21GB of memory.
[image]

When I modify the code to dynamic adjust the memory size when training, it will reduce the usage of memory as following figure. 
But unfortunately, it still face the same problem.
[image]
[image]

Thanks for replying, but I still have couple questions here.
1. What do you mean the settings.json is incorrect? Is it possible to start the AirSimNH environment if the settings.json is incorrect?

2. I currently use the official AirSimNH environment. Could it be a conflict between AirSimNH environment and local device(ubuntu 20.04 + nvidia driver 455.45.01 + cuda 11.0 + cudnn 8.0.4)?

3. I found the distance information captured by distance sensor is incorrect. It can only measure the distance in front of drone, despite the setting.json file has been modified.

wangwwno1, 2020-12-15:
> Hi @wangwwno1 ,
>
> I set the fixed GPU memory size when running the training code, so it occupy the 21GB of memory.
> [image]
>
> When I modify the code to dynamic adjust the memory size when training, it [...]

1.  Sorry for giving misleading infos, I have checked the error msg and I think this might not your case. Still, incorrect in format (e.g. additionall comma in the end of config) or value DOES cause the original AirSim environment crash during the initialization. 
2.  I'm not familiar with the AirSimNH so I'm not sure about that.
3.  This issue might related to the direction of sensors. The `Roll`, `Pitch` and `Yaw` setting in other sensors are all set to 0, that means the sensor will measure the **FRONT** direction of the drone. Try it with different values.

closed 2021-10-19 by jonyMarino