microsoft/AirSim #3203: How to force the drone flying with the specific velocity and Eular angle？

misakabribri, 2020-12-10:
OS: Windows 10
IDE: VS 2019;
Unreal: UE 4.24.3

Hello. I already have the fligt data (including positions and Euler angles) gathered from other simulation platforms. I want to use AirSim to 'playback' the flight data (just like a 3D-film) and capture images of the drone.

I guess there are at least two things to do:
1. disable the simple fight model;
2. write a new API funtion to command the drone move with the specific velocity and Eular angle;
but I don't know how to do these things.

ahmed-elsaharti, 2020-12-10:
Hi @misakabribri , this will mainly depend on the flight data you gathered. For example, if its a PX4 mission, you can simply run AirSim in [PX4 SITL or HITL mode](https://microsoft.github.io/AirSim/px4_setup/) and upload your mission to be executed.

To address the points you mentioned, if I understand what you're trying to do correctly,
1. I don't think there is a way to disable to have the drone spawn in AirSim without an autopilot mode selected. ie: you can get rid of simpleflight if you want, But you will need to select another [vehicleSetting](https://github.com/microsoft/AirSim/blob/master/docs/settings.md#vehicles-settings) if you want to spawn a UAV within Airsim (assuming you're in [Multirotor mode](https://github.com/microsoft/AirSim/blob/master/docs/settings.md#simmode)) which means that there will have to be some sort of autopilot on there (SimpleFlight, PX4Multirotor, ArduCopter or ArduRover).
2. There currently is multiple [API](https://microsoft.github.io/AirSim/apis/) functions that would let you set the moving velocity of a drone in any direction with reference to the world frame ([moveByVelocityAsync](https://github.com/microsoft/AirSim/blob/b8a06accd1f5e50d77eb10d7046042b147c0da90/PythonClient/airsim/client.py#L860-L868)) or the drone's frame ([moveByVelocityBodyFrameAsync](https://github.com/microsoft/AirSim/blob/b8a06accd1f5e50d77eb10d7046042b147c0da90/PythonClient/airsim/client.py#L838-L847)). This is done using the physics simulation and the autopilot though rather than just spawning the drone based on flight data.

misakabribri, 2020-12-14:
@ahmed-elsaharti 

Thanks! The flight data is gathered from the fixed-wing aricraft, therefore the mission can't be executed via multirotor mode on AirSim. I kown it sounds weird, but I am also trying to modify the 3D model (use the fixed-wing model instead of the default multirotor).

I found some API functions in RpcLibClientBase.hpp:
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
    // sensor omniscient APIs
    vector<int> simGetLidarSegmentation(const std::string& lidar_name = "", const std::string& vehicle_name = "") const;

    Pose simGetVehiclePose(const std::string& vehicle_name = "") const;
    void simSetVehiclePose(const Pose& pose, bool ignore_collision, const std::string& vehicle_name = "");
    void simSetTraceLine(const std::vector<float>& color_rgba, float thickness=3.0f, const std::string& vehicle_name = "");
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
1. Can I treat the vehicle as a "sensor" and change its position mannully using the fucniton simSetVehiclePose()?
2. The ComputerVision Mode won't use any phisics and vehicles. If I want to use the CV mode to get rid of the phisc engine; can I use the CV mode? (maybe it is not a good idea)

ahmed-elsaharti, 2020-12-16:
@misakabribri you could try using simSetVehiclePose to teleport the vehicle along a path but it would still be using multirotor physics.

With CV mode you pretty much just have a camera moving around in the scene.

Have a look at #2955, my understanding is that the author is working on a fixed wing model there

jonyMarino, 2020-12-22:
@ahmed-elsaharti, do you think that using CV mode and using the blueprint drone (that can be added through the ue editor) can be a way of accomplishing this? Then we can move the drone through `simSetObjectPose()`. I think this is an excellent use case for easy integration with other physics simulators.

ahmed-elsaharti, 2020-12-22:
@jonyMarino This is a BRILLIANT idea! This could also be used as a hacky solution for #3208
I just tested it and it works perfectly. Using the UE4 Editor i placed a random object (in this case a drone called BP_FlyingPawn_5, any fixed-wing model could be used too)
[image]
[image]
@misakabribri  note that the drone mesh I placed in here could be replaced with any mesh/drone
on executing:
```python
import airsim
client = airsim.VehicleClient()
client.confirmConnection()
pose1 = client.simGetObjectPose("BP_FlyingPawn_5");
pose1.position = pose1.position + airsim.Vector3r(-2, -2, -2)
success = client.simSetObjectPose("BP_FlyingPawn_5", pose1, True);
```
We get:
[image]

saihv, 2020-12-22:
The preferred way is to use CV mode and `simSetVehiclePose()`, and this is also the way we can integrate external physics engines etc. `simSetObjectPose` (while it may sometimes work), is generally not recommended for the vehicles because it targets not the physics body but more of an abstract Unreal object type. I am not sure about the following: but there might also be issue with what's considered as origin when applying `simSetObjectPose` to the vehicle itself: because the UE objects are generally referenced with the vehicle as the origin, and moving the vehicle's body to another spot may not transform the origin accurately.

That said, there is currently an open issue with `simSetVehiclePose`, where calling it even just once disables physics entirely (https://github.com/microsoft/AirSim/pull/2324 should fix it). 

The interesting next step is to go one level beyond setting the vehicle pose, and instead overriding the entire state. Currently, `simSetVehiclePose` internally calls `kinematics_->setPose(pose)`. There is also a `kinematics_->setState` that can affect the entirety of the state: pose, twist and accelerations too. https://github.com/microsoft/AirSim/blob/f423d9f5a4a0d0f4fb9a2e4ddb75251040e88f81/AirLib/include/physics/Kinematics.hpp#L13

ahmed-elsaharti, 2020-12-22:
@saihv I haven't experimented with CV mode so far so I might (will probably) be completely wrong here, but doesn't CV mode not spawn any vehicles? My understanding at this point is that the only vehicle that we can set the pose of in CV mode is the actual camera. right?

saihv, 2020-12-22:
Ah, sorry, I made a typo. I meant non-CV mode (Multirotor) and `simSetVehiclePose` is preferred for simple visualization purposes (position and orientation) instead of `simSetObjectPose`. Teleporting the drone this way seems to disable physics forever, so this might actually be a good thing if we want to continue visualizing. To actually command velocities and higher order terms, `setState` might be interesting.

ahmed-elsaharti, 2020-12-22:
@saihv Oh, no worries it makes sense now and I completely agree 👍. I was actually going to test moving the camera around in Non-CV mode to see if it messes up with anything in an attempt to see if I could create a 'fixed position' camera that tracks the drone as is needed for #3208

ahmed-elsaharti, 2020-12-22:
@saihv I've tested the CV mode + `simSetObjectPose` side of things and it does seem like the UE origin does not change once created during initialization of the simulation (*insert my regular 'I'm probably wrong' here* 😂)

What I did to test this was to set up a scene with a drone_pawn as an object in there:
[image]

This is the initial camera pose and object location:
[image]

I then grabbed the object's location and moved it 4m backwards using
```python
pose1 = client.simGetObjectPose("BP_FlyingPawn_5");
pose1.position = pose1.position + airsim.Vector3r(-4, 0, 0)
 success = client.simSetObjectPose("BP_FlyingPawn_5", pose1, True);
```
[image]

I then moved the camera backwards an equal amount
```python
camera_pose = airsim.Pose(airsim.Vector3r(-4, 0, 0), airsim.to_quaternion(0, 0, 0))
client.simSetCameraPose("0", camera_pose)
```
This resulted in only the camera moving within the original frame.
[image]

Just to make sure that everything is still referenced to that coordinate system i moved the drone backwards again to pose2:
[image]
and compared both pose1 and pose2:
[image]

which showed that they're both 4m apart with pose1 grabbed/changed **before** camera movement while pose2 was grabbed/changed **after** it. which leads me to conclude that the camera movement did not affect the coordinate system the drone/object is referenced to or the rest of the scene objects.

Is it safe to assume that this is a viable solution or is my testing method flawed?

saihv, 2020-12-22:
@ahmed-elsaharti Thanks for the quick test! `simSetCameraPose` should also just move the camera relative to an invisible vehicle (imagine an infinitely long camera arm, essentially, whereas the invisible origin stays constant). This is certainly a way to get around the origin problem. With regards to your last comment, did you mean moving the camera with the M key?

ahmed-elsaharti, 2020-12-22:
@saihv Yup, I meant the manual WASD camera controls available through the view mode enabled through the M key.

Also, would you recommend using this method to programmatically set the orientation of a fixed-location camera trying to follow a drone (or any object really) being moved around the scene to simulate a 'fixed location ground observer' scenario?

saihv, 2020-12-23:
> Also, would you recommend using this method to programmatically set the orientation of a fixed-location camera trying to follow track a drone (or any object really) being moved around the scene to simulate a 'fixed location ground observer' scenario?

Yeah actually I think it should work. I forgot about the `simSetCameraPose` function entirely (although I originally wrote that function :D) and was thinking only about simSetVehiclePose, but regardless, set camera pose should safely let you operate local to the computer vision pawn origin anyway. I am assuming in this case your drone is also just behaving as an object (and not an actual FlyingPawn vehicle)

ahmed-elsaharti, 2020-12-23:
> I forgot about the `simSetCameraPose` function entirely (although I originally wrote that function :D)

I honestly do not blame you. AirSim's got ALOT of functions that are very similarly named. This would be a good segue into the 'API documentation' topic but thats out of the issue's scope anyway 😂.

> I am assuming in this case your drone is also just behaving as an object (and not an actual FlyingPawn vehicle)

Yup for the case of #3208 my understanding is that AirSim would be used as a tool to 'visualize/draw' the data. So the drone would be an arbitrary object.

saihv, 2020-12-23:
Side note, I am noticing some weird behavior with M in CV mode. WASD + `simSetVehiclePose()` seems to work okay w.r.t. the transforms, but pressing M in CV mode does seem to destroy some camera transform somewhere because the subwindow images are going wonky.

ahmed-elsaharti, 2020-12-23:
Woops, i just realized that i meant F-mode rather than M-mode. Sorry about that

saihv, 2020-12-23:
Yeah I think as long as F / WASD works it's fine, M is probably decoupling the camera director from the pawn. WASD movement actually displaces the ComputerVisionPawn in the UE world, so that was what I was mainly curious about

closed 2021-01-14 by jonyMarino