microsoft/AirSim #3206: How to record the drone state at 1000Hz?

wangwwno1, 2020-12-11:
## Question
### What's your question?
How to record the drone state at 1000Hz?

### Include context on what you are trying to achieve
I'm trying to track the response curve of an **angular rate** controller.
Since the controller reach the target rate within 10ms, a precise plot would require shorter sampling intervals(e.g. 1ms).
So I'm wondering is there any api to change the delta time of simulation.

#### Context details
OS: Ubuntu 18.04
AirSim: master-branch(>1.3.1)
Unreal: 4.24.3

### Include details of what you already did to find answers
Before the experiment I run this code to determine the minimum simulation interval:
```
import airsim
client = airsim.MultirotorClient()
client.enableApiControl(True)
client.simPause(True)
import time
for _ in range(10):
    cur = client.getMultirotorState().timestamp
    # client.simContinueForTime(1. /1000) is not work too
    client.simPause(False)
    time.sleep(1./1000)  # 1.0 ms = 1000 Hz
    client.simPause(True)
    now = client.getMultirotorState().timestamp
    print(float(now - cur) / 10 ** 6)
```
The output of this code are either 0.0 or 3.0 ms, which is far away from my expectation.
I also check codes in `ClockBase.hpp`, `SteppableClock.hpp` and `ScalableClock.hpp` but still have no idea about how to change the simulation rate.

ahmed-elsaharti, 2020-12-11:
Hi @wangwwno1, Have you tried to change the simulation's [ClockSpeed](https://microsoft.github.io/AirSim/settings/#clockspeed)?
I would assume that slowing down the simulation would allow for more API calls per 'simulated second' (for a fixed real API call response rate).

I tried running your test code with ClockSpeed = 1 and ClockSpeed = 0.1 and there was a noticeable difference in the readings which means this might work for you.

wangwwno1, 2020-12-14:
> Hi @wangwwno1, Have you tried to change the simulation's [ClockSpeed](https://microsoft.github.io/AirSim/settings/#clockspeed)?
> I would assume that slowing down the simulation would allow for more [...]

I rewrite my code as:
```
last = client.getMultirotorState().timestamp
for _ in range(5):
    client.simContinueForTime(3. /100)  # (30ms)
    now = client.getMultirotorState().timestamp
    print(float(now - last) / 10 ** 6)
    last = now
```
And test it with ClockSpeed = 0.03333333333 (roughly as 1/30), the output is satisfied:
```
1.00096  #YAY
1.00096
1.00096
1.00096
1.00096
```

wangwwno1, 2020-12-14:
So the conclusion is:
- The time in function `simContinueForTime` is **WallClockTime**, or the elapsed time in **our world**
- What I desired is **simContinueForSimClockTime** or the time elapsed in **Simulation**
- Passing argument `secs` **shorter than 2ms** to `simContinueForTime` might make the code out of sync with simulation: the user may have an illusion that the simulation is still running(which is not)

For example, with the ClockSpeed = 0.01(the minimum interval would be 0.3ms) and time interval in `simContinueForTime` to 1ms would result in:
```
0.0
0.0
0.0
# ...... 17 0.0 omitted 
```
The code is running but no advancement in timestamp, which means the simulation is "freezed". Would be very confusing as there is no warning or exception.

Since the original question is solved, I will bring this discussion to another issue and close this one.
Thanks for your suggestion, @ahmed-elsaharti ! 👍

closed 2020-12-14 by wangwwno1